amzn/amzn-drivers #92: build fails on git download vs git checkout due to Makefile actions

chrisdag, 2018-12-23:
Trying to build this in an environment where the firewall blocks git protocol so I downloaded the master.zip only to run into this:

> fatal: Not a git repository (or any of the parent directories): .git
> (cd  && \
> 	git archive --format=tar --prefix=ena-2.0.2/ ena_linux_2.0.2 \
> 	kernel/linux/{common,ena}) > ena-2.0.2.tar
> fatal: Not a git repository (or any of the parent directories): .git
> make: *** [ena-2.0.2.tar] Error 128

Suspect this could be a documentation issue fix  (clearly document that the build will fail unless working within a git repo) or by making the Makefile agnostic to where the src code came from.

akiyano, 2018-12-23:
Hi @chrisdag,

The RPM installation assumes you are using the git version of the driver. The command that fails is simply the stripping of all the git files that are unnecessary for installation.
I have 2 suggestions for you:
1. It seems that you do have git installed on your instance (otherwise it would have complained it doesn't know what git is), so if you can clone the repository in another machine that can use the git protocol and copy the git repository to the machine you are installing on, the installation should work.

2. Or you can:
2.1. Edit the Makefile and remove the git related rows which are:
```
TOPDIR	:= $(shell git rev-parse --show-toplevel)

tarball : $(TARBALL)
$(TARBALL) : always
	(cd $(TOPDIR) && \
		git archive --format=tar --prefix=$(NAME)-$(VERSION)/ $(TAG) \
		kernel/linux/{common,ena}) > $@ 
```
2.2. Change the directory name using: mv amzn-drivers-master ena-2.0.2
2.3. Create the tar file manually (created originally by the git command): tar -cvf ena-2.0.2.tar ena-2.0.2
2.4. Move the tar file to the rpm directory: mv ena-2.0.2.tar ena-2.0.2/kernel/linux/rpm/
2.5. Continue the installation as written in the rpm readme starting again from: make

We will look into changing the script to work with non git version of the directory and/or update the documentation accordingly.

Please reply here to update if any of the above suggestions worked for you.

Thanks for the comment,
Arthur

chrisdag, 2018-12-23:
Thanks! You can close/resolve this issue as I was able to build the non-RPM version just fine from source.  I did clone the archive from a different host, tar.gz'd it and then copied it over and it still failed with the same error. Admit I did not debug that one as I really didn't need the RPM and it was very smooth to build the .ko from source

I posted the issue mainly in case you felt a README or Doc update was appropriate etc. 

Thanks!

akiyano, 2018-12-23:
OK, closing.
Thanks!

closed 2018-12-23 by akiyano