amzn/amzn-drivers #93: ena: Feature 27 isn't supported

danielmotaleite, 2019-01-02:
I have this messages in almost all my machines, using the latest ena driver (build from git)

The documentation report that snmpd network pooling is the cause, but i do not use it. After some research i found that ethtool and atop both generate this warning. atop can gather metrics every 10 min, so this is the one that is keeping this in all my systems.

So please update the documentation to include the warning not only for snmpd, but also for ethtool and atop, so other people can know where this issue is being generated.

I also opened a issue for atop to try to disable the request that generates the warning: https://github.com/Atoptool/atop/issues/46 
But the update to the documentation is still needed

danielmotaleite, 2019-01-02:
i'm talking about this documentation:
https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html

akiyano, 2019-01-02:
Hi danielmotaleite@,

The print you are referring to does not appear in any of the driver versions that you can install from this github. Please run ethtool -i <interface name - for example eth0> to see the actual installed driver version.

I will notify the writers of this documentation to make this clearer.

Thanks,
Arthur

danielmotaleite, 2019-01-02:
oops, you are right, i'm checking a group of machines that still didn't run our ena upgrade script, they are still using 1.0.2 (debian stretch).

Checking updated instances (with ena 1.6.0) i do not see that warning anymore! 
thanks for the alert!

nmeyerhans, 2019-01-02:
@danielmotaleite Future Debian stretch releases will have an up-to-date ENA driver. See [bug 915229](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=915229)

closed 2019-01-03 by akiyano