amzn/amzn-drivers #95: question: g and k in version

danielmotaleite, 2019-01-29:
Hi
just a small question, doing modinfo ena in most machines i get `2.0.2g`, but on one of then i get `2.0.2K`.
I tried to rebuild the module and its always the same. The machine with `K` is a ubuntu and most of the other machines are debian, so probably related to that.

So what is the meaning of the ending `g` and `K` ?

akiyano, 2019-01-29:
g is the driver version that you can find here in github and k is the driver that was pushed to the upstream linux kernel.

danielmotaleite, 2019-01-29:
Thank you for the info!

That was my first thought, but i'm building the driver always from github code on all machines and the previous kernel was using `g`. Recently there was a kernel update and with yesterday reboot it switched to the included kernel module, i just confirmed that it included already the latest ena, so ignoring the dkms build.

and U? i found a centos with it!  Maybe "kernel driver updated via patch"?

akiyano, 2019-01-30:
U is a deprecated letter we used in the past for releasing the driver to the Amazon Linux distribution.
CentOS probably cherry-picked it from the Amazon Linux.

closed 2019-01-30 by akiyano