amzn/amzn-drivers #98: Compile  error with gcc 8.3

thejbo, 2019-03-15:
Trying to compile ena with GCC 8.3 and Linux Kernel 5.0.2 and getting this error:

implicit declaration of function ‘dma_zalloc_coherent’; did you mean ‘dma_alloc_coherent’?

```
cat /var/lib/dkms/amzn-drivers/2.0.2/build/make.log
DKMS make.log for amzn-drivers-2.0.2 for kernel 5.0.2-gentoo (x86_64)
Fri Mar 15 01:00:05 UTC 2019
make: Entering directory '/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena'
make -C /lib/modules/5.0.2-gentoo/build M=/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena modules
make[1]: Entering directory '/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  CC [M]  /var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena/ena_netdev.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena/ena_ethtool.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena/../common/ena_com//ena_com.o
/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena/../common/ena_com//ena_com.c: In function ‘ena_com_admin_init_sq’:
/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena/../common/ena_com//ena_com.c:114:16: error: implicit declaration of function ‘dma_zalloc_coherent’; did you mean ‘dma_alloc_coherent’? [-Werror=implicit-function-declaration]
  sq->entries = dma_zalloc_coherent(queue->q_dmadev, size, &sq->dma_addr,
                ^~~~~~~~~~~~~~~~~~~
                dma_alloc_coherent
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena/../common/ena_com//ena_com.o] Error 1
make[1]: *** [Makefile:1553: _module_/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena] Error 2
make[1]: Leaving directory '/usr/src/linux-5.0.2-gentoo'
make: *** [Makefile:38: all] Error 2
make: Leaving directory '/var/lib/dkms/amzn-drivers/2.0.2/build/kernel/linux/ena'
```

Environment details:
```
System uname: Linux-5.0.0-gentoo-x86_64-AMD_Phenom-tm-_II_X6_1090T_Processor-with-gentoo-2.6
KiB Mem:    16427256 total,   5806648 free
KiB Swap:          0 total,         0 free
Timestamp of repository gentoo: Thu, 14 Mar 2019 20:30:01 +0000
sh bash 5.0_p2-r2
ld GNU ld (Gentoo 2.32 p1) 2.32.0
app-shells/bash:          5.0_p2-r2::gentoo
dev-java/java-config:     2.2.0-r4::gentoo
dev-lang/perl:            5.28.0::gentoo
dev-lang/python:          2.7.16::gentoo, 3.6.8::gentoo, 3.7.2::gentoo
dev-util/cmake:           3.13.4::gentoo
dev-util/pkgconfig:       0.29.2::gentoo
sys-apps/baselayout:      2.6-r1::gentoo
sys-apps/openrc:          0.41.2::gentoo
sys-apps/sandbox:         2.17::gentoo
sys-devel/autoconf:       2.69-r4::gentoo
sys-devel/automake:       1.13.4-r2::gentoo, 1.16.1-r1::gentoo
sys-devel/binutils:       2.32::gentoo
sys-devel/gcc:            8.3.0::gentoo
sys-devel/gcc-config:     2.0::gentoo
sys-devel/libtool:        2.4.6-r5::gentoo
sys-devel/make:           4.2.1-r4::gentoo
sys-kernel/linux-headers: 5.0-r1::gentoo (virtual/os-headers)
sys-libs/glibc:           2.28-r5::gentoo
```

AWSNB, 2019-03-15:
thanks for raising this issue.   linux 5.0 removed dma_zalloc_coherent. (https://github.com/torvalds/linux/commit/66c56cfa64d9dbb9efa8a06c1aece77e8d57ea19)

We (ENA driver team) will submit a quick patch.

meanwhile, u can try appending this to kcompat.h

#if ( LINUX_VERSION_CODE >= KERNEL_VERSION(5,0,0) )
#ifndef dma_zalloc_coherent
#define dma_zalloc_coherent(d, s, h, f) dma_alloc_coherent(d, s, h, f)
#endif

thejbo, 2019-03-15:
Thanks! I needed and extra #endif and before the final #endif:

```
#if ( LINUX_VERSION_CODE >= KERNEL_VERSION(5,0,0) )
#ifndef dma_zalloc_coherent
#define dma_zalloc_coherent(d, s, h, f) dma_alloc_coherent(d, s, h, f)
#endif
#endif

#endif /* _KCOMPAT_H_ */
```

sameehj, 2019-03-21:
Hi,

We've released a new driver here https://github.com/amzn/amzn-drivers/commit/0f03ecab2f4d7cb9b6f6998bbb0244af844ec477, which includes a fix for this issue.

Regards,
Sameeh

closed 2019-03-21 by sameehj