amzn/amzn-drivers #99: DPDK binding unbinding ENA to igb_uio

NAYANSEN90, 2019-03-15:
Hi,
So I am testing DPDK in AWS "**m5.large**" instance. The instance has 2 interfaces, both with ena drivers. I am currently using dpdk 18.05 version.  
Steps of the issue:
1. ifconfig <interface> down.
2. Bind the pci_device to igb_uio.
3. Bind the pci_device back to ena.
4. ifconfig <interface> up.

At the 4th stage the connection to the instance is lost.  I am binding and unbinding the device that doesn't contain the ssh connection. I need to reboot the system to get everything back. 

Can you help me? 
Thanks,
Nayan

NAYANSEN90, 2019-03-15:
Additional info:
The AWS instance is running OS: Centos7.4 , kernel = 3.10.0-693.21.1.el7.x86_64
I am using the "**dpdk-devbind.py**" script to bind the drivers.

AWSNB, 2019-03-15:
Thanks Nayan, just acknowledging receiving the issue, and folks looking at it

gtzalik, 2019-03-19:
Hi @NAYANSEN90 
We suspect the issue is related to network initialization scripts which is depended on your image.
We will need access to your AMI for further investigation.
Please contact me directly at gtzalik@amazon.com and provide the AMI or the source AMI you are using.

NAYANSEN90, 2019-03-20:
Hi @gtzalik ,
Thanks for the info. I have found that the problem is solved by downloading the 2.0.2g version of the driver, compiling and updating the initramfs to load the updated driver in boot time.
The above issue doesn't occur and I have been able to run dpdk  after that.
Would you say that the driver version could be the problem?

I have sent you an email with the AMI details. Hope to hear from you soon.
Thanks,
Nayan

gtzalik, 2019-03-25:
Hi @NAYANSEN90
It sounds like you used old driver version and you might hit old issue.
I'm closing this issue for now as it resolved with the latest driver. please reopen if you need anything else.
Guy

closed 2019-03-25 by gtzalik