amzn/amzn-drivers #101: rhel 7.6 ENA fails - loading out-of-tree module taints kernel

mattmartinez, 2019-04-18:
I've been trying to update our instance from t2.medium to t3.medium with an ENA supported instance type, after updating all linux packages and building/compiling/installing the latest ENA driver with the following commands:

sudo -i
cd /tmp
curl -o ena_linux_2.0.3.tar.gz https://codeload.github.com/amzn/amzn-drivers/tar.gz/ena_linux_2.0.3 
tar zxvf ena_linux_2.0.3.tar.gz
mv amzn-drivers-ena_linux_2.0.3 /usr/src/ena-2.0.3
cat <<EOF > /usr/src/ena-2.0.3/dkms.conf
PACKAGE_NAME="ena"
PACKAGE_VERSION="2.0.3"
AUTOINSTALL="yes"
REMAKE_INITRD="yes"
BUILT_MODULE_LOCATION[0]="kernel/linux/ena"
BUILT_MODULE_NAME[0]="ena"
DEST_MODULE_LOCATION[0]="/updates"
DEST_MODULE_NAME[0]="ena"
CLEAN="cd kernel/linux/ena; make clean"
MAKE="cd kernel/linux/ena; make BUILD_KERNEL=\${kernelver}"
EOF

dkms add -m ena -v 2.0.3
dkms build -m ena -v 2.0.3
dkms install -m ena -v 2.0.3
dracut -f --add-drivers ena

Afterwards the system log is showing the following error:

[    4.905752] ena: loading out-of-tree module taints kernel.
[    4.919224] ena: module verification failed: signature and/or required key missing - tainting kernel
[    4.951649] ena: Elastic Network Adapter (ENA) v2.0.3g
[    4.960649] ena 0000:00:05.0: Elastic Network Adapter (ENA) v2.0.3g
[    4.972179] ena: ena device version: 0.10
[    4.976767] ena: ena controller version: 0.0.1 implementation version 1
[    5.051209] ena 0000:00:05.0: creating 2 io queues. rx queue size: 1024 tx queue size. 1024 LLQ is ENABLED
[    5.072345] ena 0000:00:05.0: Elastic Network Adapter (ENA) found at mem febf4000, mac addr 06:7b:54:e9:2b:96 Queues 2, Placement policy: Low Latency
[   31.140099] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready

I've tried recompiling/building/installing ENA driver but to no avail, the same error " module verification failed: signature and/or required key missing - tainting kernel" continues to happen when I try to update to t3.medium.

Thanks

akiyano, 2019-04-18:
Hi @mattmartinez,

The message you are seeing is normal for installing out-of-tree drivers.
Looking at the output you provide, your driver is installed and running.
If you run ethtool -i eth0 you will see that the installed driver is indeed version 2.0.3g, which is what you installed.

Regards,
Arthur

zorikm, 2019-04-18:
@mattmartinez,
It is not clear from your post whether you have any functional issues with the driver apart of the warning messages. Also, could you please double check, if the log you provided is taken form the T3 instance?
Thanks.

mattmartinez, 2019-04-19:
> @mattmartinez,
> It is not clear from your post whether you have any functional issues with the driver apart of the warning messages. Also, could you please double check, if the log you provided is taken form the T3 instance?
> Thanks.

The log is from the system log capture from the console of the attempted t3.medium after starting it with the AMI where I installed the driver, eth0 is not coming up so I can't SSH in.

mattmartinez, 2019-04-19:
Quick update, turns out there's a specific package already in the RHEL repo that is "In tree" 

 # yum info kmod-redhat-ena.x86_64

Loaded plugins: amazon-id, rhui-lb, search-disabled-repos
Available Packages
Name        : kmod-redhat-ena
Arch        : x86_64
Version     : 2.0.2K_dup7.6
Release     : 2.el7_6
Size        : 57 k
Repo        : rhui-REGION-rhel-server-releases/7Server/x86_64
Summary     : ena module for Driver Update Program
URL         : http://www.kernel.org/ 
License     : GPLv2
Description : ena module for Driver Update Program

I installed this package and the instance came up with networking without issues, thanks for the comments.

[root@ip-172-18-101-123 ~]# ethtool -i eth0
driver: ena
version: 2.0.2K_dup7.6
firmware-version:
expansion-rom-version:
bus-info: 0000:00:05.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

closed 2019-04-19 by mattmartinez