amzn/amzn-drivers #102: DPDK on EC2 instances only Sends traffic but not receiving

Smiledarshu, 2019-04-26:
Hi,

I am new to AWS EC2, 

I am running an instance c5.xlarge on EC2 AWS, dpdk_18.05.02 is compiled and ports are binded, 
I have two private subnets/interfaces eth1 and eth2,  when i try to send traffic from(eth1-tx) i can't able to receive the same to (eth2-Rx),

My question is how to make the traffic full duplex? (TxRx-TxRx)

Please help me,
Thanks

nafeabshara, 2019-04-26:
Hi

Thanks for using EC2 and your inquiry

Could you check the network security group rules associated with both interfaces ?

Another thing to validate that each interface is using the proper interface MAC address as source MAC.  If source MaC is wrong, packets  will be filtered

Thx

Smiledarshu, 2019-04-29:
Hi nafeabshara,

Each interface is using proper MAC address as source.

"_Could you check the network security group rules associated with both interfaces ?_"
How to define security group rules if i wanted to send packets from Eth1 to Eth2 and vice-versa?

Also attached eth1 and eth2 inbound and outbound rules image.

Thanks
[image]
[image]

zorikm, 2019-04-30:
@Smiledarshu,
Could you please provide us instance-id you are using.

Smiledarshu, 2019-04-30:
@zorikm 
this is the instance id I am using,
i-06d18b901a8277bb9

Smiledarshu, 2019-04-30:
Hi All,

Anyone got anything about how to proceed further

Thanks,

segevido, 2019-04-30:
Hello Smiledarshu,

When working over different subnets you need to use the gateway MAC address as the destination MAC address and not the remote instance MAC address.
Can you please check destination MAC address as well?

We've tested the same scenario with testpmd and it works fine.
Do you use a public application or a private one?

Thanks,
Ido

Smiledarshu, 2019-04-30:
Hi @segevido evido,

We use private application.

I have two interfaces eth1(172.31.17.214) and eth2(172.31.17.20) and both are in the same subnet.
Where i can find the gateway MAC address?

segevido, 2019-04-30:
Hi @Smiledarshu,

We've checked different scenario since you mentioned in your first message that you use 2 subnets.
We will check it again with 2 interfaces in the same subnet (1 instance).

The gateway MAC address is relevant while using different subnets, so it's not your case.
Anyway, you can see your gateway in the local routing table by the command `route`.
Then, you can get the MAC address by the command `arp`.

Thanks,
Ido

Smiledarshu, 2019-04-30:
@segevido 
My `route` output,

 route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         ip-172-31-16-1. 0.0.0.0         UG    100    0        0 eth0
172.31.16.0     0.0.0.0         255.255.240.0   U     100    0        0 eth0

and my `arp`

 arp
Address                  HWtype  HWaddress           Flags Mask            Iface
ip-172-31-16-1.ap-south  ether   02:85:4d:37:73:06   C                     eth0

Thanks,

segevido, 2019-04-30:
Hi @Smiledarshu, 

We've checked it again with same setup as you use - 1 instance with 3 ENIs which are in the same subnet; 1 with the ENA Linux driver and 2 others with ENA DPDK.
We were able to send and receive traffic between the 2 DPDK interfaces.

The `route` command output is not relevant in this case since all of the ENIs are in the same subnet.

Please double check both MAC and IP addresses (both source and destination).
Make sure you use the correct ones (don't use the one from the `ifconfig` since it represents the Linux-binded driver).

Ido

Smiledarshu, 2019-05-01:
Hi @segevido ,

1) What you have tested in 2 DPDK interfaces, I mean which sample application?

2) If i use ifconfig only eth0 will appear because eth1 and eth2 are binded to DPDK. I am using Private IP's from Network Interfaces section, and  mac address are fine too.

Thanks,

segevido, 2019-05-02:
Hi @Smiledarshu,

1. We were using Pktgen Ver: 3.5.9 (DPDK 18.05.1).

2. `ifconfig` was an example of what not to use. It's good you have used the addresses which are assigned to the correct interfaces.

Please contact us by email - idose@amazon.com and gtzalik@amazon.com

gtzalik, 2019-05-20:
Hi @Smiledarshu,

I hope the issue was resolved for you as we didn't hear from you for long time
Please contact us if any further assistance is needed

closed 2019-05-20 by gtzalik