amzn/amzn-drivers #103: Issue receiving incoming packets with sockets using SO_ATTACH_REUSEPORT_CBPF

estaban, 2019-04-28:
I have a multi-threaded UDP server, all listening threads are bound to the same port using SO_REUSEPORT. I am overriding the load balancing algorithm via SO_ATTACH_REUSEPORT_CBPF, the dispatching function is simply the core number the interrupt is being processed on.

Incoming UDP datagrams larger than rx_copybreak minus the IP header are not being received by the application (tested with multiple values of rx_copybreak). However, datagrams smaller than that threshold are successfully received by the application.

I am using RHEL 5.3, and ena driver version 2.0.3g.

gtzalik, 2019-04-29:
@estaban Thanks for contacting us
We will investigate it and will let you know about our findings.

sameehj, 2019-04-29:
Hi @estaban ,

Thank you for reporting this, I'll be working on this with you for resolving the issue.

We ran some sanity checks with some packet generator to make sure this doesn't happen in each
and every scenario and indeed it didn't reproduce with udp traffic with various UDP datagram sizes.

In order to make this efficient, can you please contact me through my email address <sameehj@amazon.com> so that we continue investigating the issue offline. I need you to provide me with more information on how to reproduce this.

Thanks,
Sameeh

sameehj, 2020-08-27:
The solution for this issue is the same as the solution for https://github.com/amzn/amzn-drivers/issues/137
Therefore closing this issue as a duplicate.

closed 2020-08-27 by sameehj