amzn/amzn-drivers #105: compiled successfully but installation failed on Centos 7.6 - signature and/or required key missing - tainting kernel

daniejstriata, 2019-05-09:
Hi,

Is it possible to create better installation documents? I'm using the README from this git repo and a AWS guide found here and it does not work for me on Centos 7.6. I must be missing a step somewhere to sign the driver due to the error:
`
 kernel: ena: module verification failed: signature and/or required key missing - tainting kernel
`
https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html#enhanced-networking-ena-linux

As far as I can see the driver is installed but is not being used but I do not see anything about signing the driver or is this not the reason it is failing to load?

Message Log:

```
May  9 11:56:54 eaew2p-af2l kernel: ena: Elastic Network Adapter (ENA) v2.1.0g
May  9 13:23:03 eaew2p-af2l kernel: ena: loading out-of-tree module taints kernel.
May  9 13:23:03 eaew2p-af2l kernel: ena: module verification failed: signature and/or required key missing - tainting kernel
```

```
# cat /etc/modules-load.d/ena.conf
ena
```
Driver is installed
```
 lsmod | grep ena
ena                   102750  0 
```
And is loaded in the kernel:
` kernel: ena: Elastic Network Adapter (ENA) v2.1.0g`
```
# modinfo ena
filename:       /lib/modules/3.10.0-957.12.1.el7.x86_64/extra/ena/ena.ko
version:        2.1.0g
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
retpoline:      Y
rhelversion:    7.6
srcversion:     DF36FD877CC81A3596E9C8C
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
depends:        
vermagic:       3.10.0-957.12.1.el7.x86_64 SMP mod_unload modversions 
parm:           debug:Debug level (0=none,...,16=all) (int)
parm:           rx_queue_size:Rx queue size. The size should be a power of 2. Max value is 8K
 (int)
parm:           force_large_llq_header:Increases maximum supported header size in LLQ mode to 224 bytes, while reducing the maximum TX queue size by half.
 (int)

```
I ran depmod followed by dracut -f -v and rebooted

When I run:
```
# ethtool -i eth0
driver: vif
version: 
firmware-version: 
expansion-rom-version: 
bus-info: vif-0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```
but when I run:

zorikm, 2019-05-09:
@daniejstriata ,
Thanks for reaching out.
What instance type are you using? Can you please specify your instance-id?
Thanks.

daniejstriata, 2019-05-09:
It is:
                    "ImageId": "ami-02bd0869dd335281f",
                    "InstanceId": "i-0cab4538bbec67b0e",
                    "InstanceType": "t2.large",

daniejstriata, 2019-05-09:
@zorikm I read some more on this matter. It seams that drivers don't have to be signed. Maybe I'm on a wild goose chase to get the driver loaded.

zorikm, 2019-05-09:
@daniejstriata,
t2 instances do not use ENA.
Please refer to https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html#enhanced-networking-ena-linux where you can find list of instances using ENA.
Thanks for your suggestion to update our documentation. We will definitely do this ASAP.

daniejstriata, 2019-05-09:
@zorikm That is the documentation I used to get where I am now. I want to add ENA to the t2.large so that I can shut it down and convert it to a t3.large but without ENA you cannot do it.

zorikm, 2019-05-09:
@daniejstriata, not sure I'm following. 
t2 doesn't have ENA device. So adding the ENA driver wouldn't help, it wouldn't be loaded as the device is not present.
What do you mean by "convert to a t3.large". Are you trying to add ENA driver to the AMI on t2.large, then create a new AMI from it, and launch it on t3.large?

akiyano, 2019-05-29:
@daniejstriata,
Here is what I tried and succeeded:

First try - simple case:
I took the latest Centos 7 ami in N. Virginia (ami-02eac2c0129f6376b).
I created an instance from it on t2.large.
I saved a new ami from this instance.
I ran an instance from the new ami on t3.large. And it ran fine. This is because the original ami that I started with had ena installed and enabled in it.

Second try - more complex case:
I took the latest Centos 7 ami in N. Virginia (ami-02eac2c0129f6376b).
I created an instance from it on t2.large.
I installed the latest ena driver from github using dkms using the following steps:

sudo yum install git make gcc kernel-devel-$(uname -r)
git clone https://github.com/amzn/amzn-drivers.git

DKMS installation (unfortunately simply running sudo yum install dkms doesn't work):
sudo yum install yum-utils
curl -O https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm
yum clean metadata dbcache
sudo yum install epel-release-latest-7.noarch.rpm
sudo yum-config-manager --disable epel
sudo yum --enablerepo=epel install dkms

sudo mv amzn-drivers /usr/src/amzn-drivers-2.1.0

edit  /usr/src/amzn-drivers-2.1.0/dkms.conf using your favorite text editor and write in it:
PACKAGE_NAME="ena"
PACKAGE_VERSION="1.0.0"
CLEAN="make -C kernel/linux/ena clean"
MAKE="make -C kernel/linux/ena/ BUILD_KERNEL=${kernelver}"
BUILT_MODULE_NAME[0]="ena"
BUILT_MODULE_LOCATION="kernel/linux/ena"
DEST_MODULE_LOCATION[0]="/updates"
DEST_MODULE_NAME[0]="ena"
AUTOINSTALL="yes"

sudo dkms add -m amzn-drivers -v 2.1.0
sudo dkms build -m amzn-drivers -v 2.1.0
sudo dkms install -m amzn-drivers -v 2.1.0

At this point driver 2.1.0 is installed on the instance.

I saved a new ami from this instance.
I ran an instance from the new ami on t3.large, it ran fine and ethtool -i showed me that the installed driver is 2.1.0.

What might have happened in your case was that your original ami was not ena enabled. I suggest you follow the instructions [here](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html) to enable ena in the ami and see if it fixes your problem.

In case you are still having problems with this, please contact me via email: akiyano@amazon.com

P.S.
the kernel tainting + driver signature failing messages are not a problem, they simply say you installed a driver from an outside source that is not created by CentOS.

closed 2019-05-29 by akiyano