amzn/amzn-drivers #108: how to load bnx driver in dpdk for NetXtreme II BCM5709 NIC's

Smiledarshu, 2019-06-12:
Hi all,

I have Dell R610 server with two Broadcom Netxtreme II BCM5709 NIC card,

When I am running the test application it's showing the following:

"EAL: Detected 8 lcore(s)
EAL: Detected 1 NUMA nodes
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: No free hugepages reported in hugepages-1048576kB
EAL: Probing VFIO support...
1.0              lcores: RX( 1 )TX( 1 ) ports: RX( 0 )TX( 0 )
2.1              lcores: RX( 2 )TX( 2 ) ports: RX( 1 )TX( 1 )
*** Did not find any ports to use ***
=== port to lcore mapping table (# lcores 4) ===
   lcore:     0     1     2     3 
port   0:  D: T  1: 1  0: 0  0: 0 =  1: 1
port   1:  D: T  0: 0  1: 1  0: 0 =  1: 1
Total   :  0: 0  1: 1  1: 1  0: 0
    Display and Timer on lcore 0, rx:tx counts per port/lcore

Configuring 0 ports, MBUF Size 1920, MBUF Cache Size 1024Lcore:
    1, RX-TX  
                RX( 1): ( 0: 0) 
                TX( 1): ( 0: 0) 
    2, RX-TX  
                RX( 1): ( 1: 0) 
                TX( 1): ( 1: 0) 

Port :
    0, nb_lcores  1, private (nil), lcores:  1 
    1, nb_lcores  1, private (nil), lcores:  2 "

Please help me to solve this error

thanks

akiyano, 2019-06-13:
Hi @Smiledarshu,

Could you please explain the connection of your question to ENA?

Thanks,
Arthur

Smiledarshu, 2019-06-14:
Hi @akiyano,

Actually, I am trying to run a dpdk application using NetXtreme II BCM5709 NIC cards, 

After binding ports to dpdk when I run the test application it is giving error like "Did not find any ports".

I have enabled CONFIG_RTE_LIBRTE_BNXT_PMD=y in common_base file.

Can you help me?

Thanks

akiyano, 2019-06-17:
Hi @Smiledarshu,

I’m afraid you are asking your question in the wrong place.
In this repository we deal with ENA drivers, we can not provide support for DPDK issues not related to ENA.

Thank You For Your Understanding,
Arthur

closed 2019-06-17 by akiyano