amzn/amzn-drivers #109: CRC32 RSS Hash uses non-standard polynomial?

ashaffer, 2019-06-27:
I'm attempting to get [seastar](https://github.com/scylladb/seastar) to work with DPDK on my EC2 instance. It's almost there, but I can't get RSS to work properly. The first issue is that I can't just tell it to use Toeplitz hashing per #80. The second issue though is that even when I try to do CRC32 hashing for RSS like the docs here claim it's doing...the hashes don't match. In fact, the value i'm getting tends to have its upper and lower word mirrored, e.g.:

`0x13541354`

Any idea what's going on here?

ashaffer, 2019-06-30:
@netanelbelgazal bump. Am I doing something wrong, because it seems like the RSS hash i'm getting is clearly neither CRC32 nor Toeplitz.

akiyano, 2019-07-01:
Hi @ashaffer,
This issue is not dpdk specific, it is a hardware feature that needs to be enabled, which is why it is actually the same as https://github.com/amzn/amzn-drivers/issues/80
Closing this issue as duplicate of https://github.com/amzn/amzn-drivers/issues/80
Regards,
Arthur

closed 2019-07-01 by akiyano

ashaffer, 2019-07-01:
@akiyano It's not a duplicate. I'm not asking if I can configure the hash, i'm asking what the current hash is. In order to take advantage of core locality properly, I need to be able to predict which CPU an incoming packet is going to land on, but upon examining the RSS hash the ENA is actually generating, it doesn't appear to be Toeplitz or CRC32.

zorikm, 2019-07-01:
Hi @ashaffer,
Apologies for some confusion. 
On the instance type you use, the device utilizes Toeplitz and the resulted hash value is 16b. It is duplicated to make it 32 bit.
Hope this helps. Please let us know if you have further questions.

ashaffer, 2019-07-01:
Can you tell me the hash key it's using? Is it the mellanox one that people seem to use?

zorikm, 2019-07-01:
The one used by majority of the NIC vendors:
0x6d5a56da, 0x255b0ec2, 0x4167253d, 0x43a38fb0, 0xd0ca2bcb, 0xae7b30b4, 0x77cb2da3, 0x8030f20c, 0x6a42b73b, 0xbeac01fa

ashaffer, 2019-07-01:
Great, thanks so much :)

ashaffer, 2019-07-02:
@zorikm Sorry if i'm being dense here, but i'm not able to get the standard Toeplitz hash to work with that key. I notice that you said the resultant key is 16 bits, but the standard Toeplitz hash i'm familiar with generates a 32 bit output. Can you point me to the Toeplitz implementation you're using?

zorikm, 2019-07-02:
@ashaffer. We'll be happy to help you. Please reach out to me (zorik@amazon.com)

soulofmitsui, 2021-04-02:
Hi,
I'm using DPDK on ENA 2.4.0g. The rss hash returned by ENA doesn't math Toeplitz. Can you please confirm the hash key ENA uses is {0x6d5a56da, 0x255b0ec2, 0x4167253d, 0x43a38fb0, 0xd0ca2bcb, 0xae7b30b4, 0x77cb2da3, 0x8030f20c, 0x6a42b73b, 0xbeac01fa}? What's the initial hash value if not 0?

I-gor-C, 2021-04-02:
@soulofmitsui Please see answer @ https://github.com/amzn/amzn-drivers/issues/169