amzn/amzn-drivers #110: DPDK ENA PMD gives only 10Gbps performance on a m4 16x large instance.

networkbull, 2019-08-16:
I am observing that the bandwidth for ENA PMD gets limited to 10 Gbps when I am using TestPMD (DPDK 19.05) application with 1 ENA interface in m4 16x Large instance. I am running traffic between a client and server but the traffic seems to be limited to 10Gpbs.

If I perform the same test with ENA Kernel Driver from the instance, I get 25Gpbs ouput using iperf client and server.

Can you please let me know what can be the issue in DPDK ENA PMD which leads to such a behaviour.

nafeabshara, 2019-08-16:
How many concurrent flows are you running in dpdk case ?     

Typically a single flow is limited to 10G and need to run multiple flows to get to 25

networkbull, 2019-08-16:
I tried with 4 concurrent flows with changing port numbers.
From the ENA PMD the link speed that gets displayed is 0 Mbps, in that case how can we know the current link speed that gets negotiated.

networkbull, 2019-08-18:
From the ENA PMD it seems, ENA driver is applying link->speed as 0, because of which any Test Application like ( PktGen ) defaults to 10G.

I understand that the speed of the interface has been disabled via ethtool, but ena_com_get_link_params can be used to get the Admin Speed, Supported Params and Mode ( AutoNeg, Duplex ), however it returns error as -95. 

Is error number -95 specific error while communicating with Admin.

I-gor-C, 2019-08-19:
@networkbull The link speed isn't reported on purpose (due to cloud's environment where link speed is not fixed).

networkbull, 2019-08-19:
Thanks @I-gor-C, I am observing that my perf remains at 10Gbps even when I am using m4_16x large instances with two interfaces/VM ( 1 Management, 1 Data ).

I-gor-C, 2019-08-19:
@networkbull Like you said - the test app defaults to 10Gbps when linkspeed isn't reported. The speed is per interface, number of interfaces is unrelated. You can run multiple flows in parallel through one interface and see that the total throughput is higher.

networkbull, 2019-08-19:
@I-gor-C I modified the test app to use 25Gbps linkspeed, even with multiple flows the bandwidth for Amazon instance doesnot go beyond 10Gbps. I am using packet-gen application on both server and client side.

semihalf-krawczyk-michal, 2019-08-19:
Hi @networkbull,

I've just checked with our internal tool for testing DPDK performance on m4.16xl and I can achieve 25Gb/s using 8 flows of 9000B packets and above 24Gb/s using 8 flows of 1500B packets. I'm using 8 parallel DPDK queues each with different flow. Could you, please, tell me, how are you setting up the pkt-gen and running the flow and how did you changed the link speed in the pktgen?

networkbull, 2019-08-19:
Hi @semihalf-krawczyk-michal ,

Please find below patch for pkt-gen,
`
diff --git a/app/pktgen-port-cfg.c b/app/pktgen-port-cfg.c
index 4b1bd18..d756887 100644
--- a/app/pktgen-port-cfg.c
+++ b/app/pktgen-port-cfg.c
@@ -409,7 +409,8 @@ pktgen_config_ports(void)
 #endif
 			txconf->offloads = default_port_conf.txmode.offloads;
 
-			ret = rte_eth_tx_queue_setup(pid, q, pktgen.nb_txd, sid, txconf);
+			//ret = rte_eth_tx_queue_setup(pid, q, pktgen.nb_txd, sid, txconf);
+			ret = rte_eth_tx_queue_setup(pid, q, 1024, sid, txconf);
 			if (ret < 0)
 				pktgen_log_panic("rte_eth_tx_queue_setup: err=%d, port=%d, %s",
 						 ret, pid, rte_strerror(-ret));
diff --git a/app/pktgen-stats.c b/app/pktgen-stats.c
index 9b97ce0..55bf35e 100644
--- a/app/pktgen-stats.c
+++ b/app/pktgen-stats.c
@@ -243,7 +243,6 @@ pktgen_get_link_status(port_info_t *info, int pid, int wait)
 {
 	int i;
 	uint64_t prev_status = info->link.link_status;
-
 	/* get link status */
 	for (i = 0; i < LINK_RETRY; i++) {
 		memset(&info->link, 0, sizeof(info->link));
@@ -263,7 +262,7 @@ pktgen_get_link_status(port_info_t *info, int pid, int wait)
 
 	/* Setup a few default values to prevent problems later. */
 #if RTE_VERSION >= RTE_VERSION_NUM(17,2,0,0)
-	info->link.link_speed   = ETH_SPEED_NUM_10G;
+	info->link.link_speed   = ETH_SPEED_NUM_25G;
 #else
 	info->link.link_speed   = 10000;
 #endif
`
Setup Topology is launching two instances of m4.16xl on the same VPC. 

**Procedure:**
1. Build packet-gen on both the server and client instances using pkt-gen.
2. Make sure all the pre-requisite are done for configuring DPDK enviornment,
3. Start the pkt-gen application as shown below,
# ./app/x86_64-native-linuxapp-gcc/pktgen -l 0-16 -n 4 -- -P -m "[1-8:9-16].0" -N
4. Configure the client and server ip address, mac address and range of ports. 
5. Start the Application on client and check data on server.

Thanks

semihalf-krawczyk-michal, 2019-08-20:
I was able to reproduce your issue only if there was single flow running from the Tx.

Please make sure that you are enabling the range verify using tcpdump on the Rx machine  on second interface binded to the kernel that in fact you are sending multiple flows from the Tx.

To enable 'range' for port 0, you must pass `enable 0 range` inside the pktgen.

Also you can verify that on Rx only single queue is working, by passing `page stats` - to enable multiple Rx queues, you could add a patch like this:

```
diff --git a/app/pktgen-port-cfg.c b/app/pktgen-port-cfg.c
index 4b1bd18..78ed482 100644
--- a/app/pktgen-port-cfg.c
+++ b/app/pktgen-port-cfg.c
@@ -59,6 +59,7 @@ static struct rte_eth_conf default_port_conf = {
        },
 #else
        .rxmode = {
+               .mq_mode = ETH_MQ_RX_RSS,
                .split_hdr_size = 0,
 #if RTE_VERSION < RTE_VERSION_NUM(18, 11, 0, 0)
                .offloads = DEV_RX_OFFLOAD_CRC_STRIP,
```

After doing those 2 steps you should be to get more than 10Gb/s. Pktgen doesn't allow to create jumbo frames, so if you would like to use bigger packets, please apply patch like this:

```
diff --git a/app/pktgen.h b/app/pktgen.h
index 3b79848..f7e895a 100644
--- a/app/pktgen.h
+++ b/app/pktgen.h
@@ -234,7 +234,7 @@ enum {
                                   PKT_PREAMBLE_SIZE + ETHER_CRC_LEN),

    MIN_PKT_SIZE            = (ETHER_MIN_LEN - ETHER_CRC_LEN),
-   MAX_PKT_SIZE            = (ETHER_MAX_LEN - ETHER_CRC_LEN),
+   MAX_PKT_SIZE            = (9000 - ETHER_CRC_LEN),
    MIN_v6_PKT_SIZE         = (78 - ETHER_CRC_LEN),

    MAX_RX_QUEUES           = 16,   /**< RX Queues per port */
```
Thanks,
Michal

networkbull, 2019-08-30:
Thanks @I-gor-C , @semihalf-krawczyk-michal  the issue cam up from the Rx Que and Tx Que allocation mechanism from pkt-gen, it was not utilising all the queues while sending our receiving traffic. 
After further modifications I am able to see line rate performance with 8 flows on instances.
Closing the issue with this comment.

closed 2019-08-30 by networkbull