amzn/amzn-drivers #111: Unable to run 'make' for FreeBSD 11.3

nonamef, 2019-09-19:
I followed the steps to'make' the ena driver for FreeBSD 11.3, but it fails with the following issue.
```
Warning: Object directory not changed from original /tmp/amzn-drivers-master/kernel/fbsd/ena
cc -O2 -pipe  -DENA_DEBUG -fno-strict-aliasing -Werror -D_KERNEL -DKLD_MODULE -nostdinc   -I. -I/usr/src/sys -fno-common  -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer   -MD  -MF.depend.ena.o -MTena.o -mcmodel=kernel -mno-red-zone -mno-mmx -mno-sse -msoft-float  -fno-asynchronous-unwind-tables -ffreestanding -fwrapv -fstack-protector -Wall -Wredundant-decls -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Winline -Wcast-qual -Wundef -Wno-pointer-sign -D__printf__=__freebsd_kprintf__ -Wmissing-include-dirs -fdiagnostics-show-option -Wno-unknown-pragmas -Wno-error-tautological-compare -Wno-error-empty-body -Wno-error-parentheses-equality -Wno-error-unused-function -Wno-error-pointer-sign -Wno-error-shift-negative-value -Wno-error-address-of-packed-member  -mno-aes -mno-avx  -std=iso9899:1999 -c ena.c -o ena.o
ena.c:4661:38: error: too few arguments provided to function-like macro invocation
    nitems(ena_vendor_info_array) - 1);
                                     ^
/usr/src/sys/sys/module.h:172:9: note: macro 'MODULE_PNP_INFO' defined here
#define MODULE_PNP_INFO(d, b, unique, t, l, n)                          \
        ^
ena.c:4660:1: error: type specifier missing, defaults to 'int' [-Werror,-Wimplicit-int]
MODULE_PNP_INFO("U16:vendor;U16:device", pci, ena, ena_vendor_info_array,
^
2 errors generated.
*** Error code 1

Stop.
make: stopped in /tmp/amzn-drivers-master/kernel/fbsd/ena
```
Need to generate this driver as the version 'v0.8.4' installed with FreeBSD 11.3 is failing every few days for my r4.xlarge EC2.

segevido, 2019-09-19:
Hello @nonamef ,

We will make sure to fix this soon. I will update here as well.

Thanks,
Ido

segevido, 2019-09-19:
Hello @nonamef,

Please find the following patch for fixing you issue.
Next, we will merge the patch to 'master' branch.

Please let me know if you have any more issues.

Thanks,
Ido
[0001-fbsd-ena-fix-MODULE_PNP_INFO-call-for-FreeBSD-11.zip](https://github.com/amzn/amzn-drivers/files/3631316/0001-fbsd-ena-fix-MODULE_PNP_INFO-call-for-FreeBSD-11.zip)

nonamef, 2019-09-20:
Worked like a charm. Thanks for the awesome response time.

closed 2019-09-20 by nonamef