amzn/amzn-drivers #117: DKMS: amzn-drivers-ena-linux/1.1.3 compilation error (Debian 10 / 4.19.0-6-amd64)

KodyWiremane, 2019-11-13:
Hello, I've upgraded to a newer kernel today, and in APT's output I noticed unusual output:
```
Setting up linux-compiler-gcc-8-x86 (4.19.67-2+deb10u2) ...
Setting up linux-kbuild-4.19 (4.19.67-2+deb10u2) ...
Setting up linux-image-4.19.0-6-amd64 (4.19.67-2+deb10u2) ...
/etc/kernel/postinst.d/dkms:
Error! Bad return status for module build on kernel: 4.19.0-6-amd64 (x86_64)
Consult /var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/make.log for more information.
Error! Bad return status for module build on kernel: 4.19.0-6-amd64 (x86_64)
Consult /var/lib/dkms/ixgbevf/3.2.2/build/make.log for more information.
```

Here are the log files mentioned:
<details><summary>amzn-drivers-ena-linux's</summary>

```
DKMS make.log for amzn-drivers-ena-linux-1.1.3 for kernel 4.19.0-6-amd64 (x86_64)
Wed 13 Nov 2019 04:24:31 PM UTC
make: Entering directory '/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena'
make -C /lib/modules/4.19.0-6-amd64/build M=/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena modules
make[1]: Entering directory '/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  CC [M]  /var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena/ena_netdev.h: In function ‘ena_bp_unlock_napi’:
/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena/ena_netdev.h:387:21: error: ‘struct napi_struct’ has no member named ‘gro_list’; did you mean ‘gro_hash’?
  if (rx_ring->napi->gro_list)
                     ^~~~~~~~
                     gro_hash
compilation terminated due to -Wfatal-errors.
make[4]: *** [/usr/src/linux-headers-4.19.0-6-common/scripts/Makefile.build:309: /var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena/ena_netdev.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-6-common/Makefile:1534: _module_/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-6-amd64'
make: *** [Makefile:19: all] Error 2
make: Leaving directory '/var/lib/dkms/amzn-drivers-ena-linux/1.1.3/build/kernel/linux/ena'
```
</details>

<details><summary>ixgbevf's</summary>

```
DKMS make.log for ixgbevf-3.2.2 for kernel 4.19.0-6-amd64 (x86_64)
Wed 13 Nov 2019 04:24:33 PM UTC
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-6-common'
make[2]: Entering directory '/usr/src/linux-headers-4.19.0-6-amd64'
  CC [M]  /var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.o
In file included from /var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbe_vf.h:232,
                 from /var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf.h:45,
                 from /var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:55:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_osdep2.h: In function ‘IXGBE_WRITE_REG’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_osdep2.h:34:13: error: implicit declaration of function ‘ACCESS_ONCE’; did you mean ‘READ_ONCE’? [-Werror=implicit-function-declaratio
n]
  reg_addr = ACCESS_ONCE(hw->hw_addr);
             ^~~~~~~~~~~
             READ_ONCE
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_osdep2.h:34:11: warning: assignment to ‘u8 *’ {aka ‘unsigned char *’} from ‘int’ makes pointer from integer without a cast [-Wint-conv
ersion]
  reg_addr = ACCESS_ONCE(hw->hw_addr);
           ^
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_osdep2.h: In function ‘IXGBE_WRITE_REG64’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_osdep2.h:58:11: warning: assignment to ‘u8 *’ {aka ‘unsigned char *’} from ‘int’ makes pointer from integer without a cast [-Wint-conv
ersion]
  reg_addr = ACCESS_ONCE(hw->hw_addr);
           ^
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: In function ‘ixgbevf_validate_register_read’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:184:11: warning: assignment to ‘u8 *’ {aka ‘unsigned char *’} from ‘int’ makes pointer from integer without a cast [-Wint-conve
rsion]
  reg_addr = ACCESS_ONCE(hw->hw_addr);
           ^
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: In function ‘ixgbe_read_reg’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:205:11: warning: assignment to ‘u8 *’ {aka ‘unsigned char *’} from ‘int’ makes pointer from integer without a cast [-Wint-conve
rsion]
  reg_addr = ACCESS_ONCE(hw->hw_addr);
           ^
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: In function ‘ixgbevf_busy_poll_recv’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:1322:10: error: ‘LL_FLUSH_FAILED’ undeclared (first use in this function); did you mean ‘MF_FAILED’?
   return LL_FLUSH_FAILED;
          ^~~~~~~~~~~~~~~
          MF_FAILED
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:1322:10: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:1325:10: error: ‘LL_FLUSH_BUSY’ undeclared (first use in this function); did you mean ‘TLB_FLUSH_VMA’?
   return LL_FLUSH_BUSY;
          ^~~~~~~~~~~~~
          TLB_FLUSH_VMA
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: In function ‘ixgbevf_acquire_msix_vectors’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:2604:9: error: implicit declaration of function ‘pci_enable_msix’; did you mean ‘pci_enable_msi’? [-Werror=implicit-function-de
claration]
   err = pci_enable_msix(adapter->pdev, adapter->msix_entries,
         ^~~~~~~~~~~~~~~
         pci_enable_msi
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: In function ‘ixgbevf_alloc_q_vector’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:2821:2: error: implicit declaration of function ‘napi_hash_add’; did you mean ‘napi_hash_del’? [-Werror=implicit-function-decla
ration]
  napi_hash_add(&q_vector->napi);
  ^~~~~~~~~~~~~
  napi_hash_del
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: At top level:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:4459:21: error: initialization of ‘void (*)(struct net_device *, struct rtnl_link_stats64 *)’ from incompatible pointer type ‘s
truct rtnl_link_stats64 * (*)(struct net_device *, struct rtnl_link_stats64 *)’ [-Werror=incompatible-pointer-types]
  .ndo_get_stats64 = ixgbevf_get_stats64,
                     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:4459:21: note: (near initialization for ‘ixgbevf_netdev_ops.ndo_get_stats64’)
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:4479:3: error: ‘const struct net_device_ops’ has no member named ‘ndo_busy_poll’
  .ndo_busy_poll  = ixgbevf_busy_poll_recv,
   ^~~~~~~~~~~~~
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:4479:20: error: initialization of ‘void (*)(struct net_device *)’ from incompatible pointer type ‘int (*)(struct napi_struct *)
’ [-Werror=incompatible-pointer-types]
  .ndo_busy_poll  = ixgbevf_busy_poll_recv,
                    ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:4479:20: note: (near initialization for ‘ixgbevf_netdev_ops.ndo_poll_controller’)
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c: In function ‘ixgbevf_probe’:
/var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.c:4722:2: error: implicit declaration of function ‘setup_timer’; did you mean ‘sk_stop_timer’? [-Werror=implicit-function-declara
tion]
  setup_timer(&adapter->service_timer, &ixgbevf_service_timer,
  ^~~~~~~~~~~
  sk_stop_timer
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-4.19.0-6-common/scripts/Makefile.build:309: /var/lib/dkms/ixgbevf/3.2.2/build/src/ixgbevf_main.o] Error 1
make[2]: *** [/usr/src/linux-headers-4.19.0-6-common/Makefile:1534: _module_/var/lib/dkms/ixgbevf/3.2.2/build/src] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.19.0-6-amd64'
make[1]: *** [Makefile:146: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-6-common'
make: *** [Makefile:71: default] Error 2
```
</details>

Evolving from AMI ID `debian-jessie-amd64-hvm-2017-01-15-1221-ebs (ami-b14ba7a7)`, now it's
```
Debian GNU/Linux 10 (buster)
Linux machine 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64 GNU/Linux
```
APT's sources:
```
deb https://cloudfront.debian.net/debian stable main contrib non-free
deb https://cloudfront.debian.net/debian-security stable/updates main contrib non-free
deb https://cloudfront.debian.net/debian stable-updates main contrib non-free
```

I made it boot with the previous kernel by default and made a snapshot, just in case. Now I switched it to the new kernel, and it works, at least I have network access to the instance. Though I don't know much about DKMS, so I can't say what exactly my OS has loaded.

Any advice on the situation?

sameehj, 2019-11-14:
Hi @KodyWiremane ,

Thank you for reporting this, I'll be working on this with you for resolving the issue.

In order to make this efficient, can you please contact me through my email address sameehj@amazon.com
so that we continue investigating the issue offline.

Thanks,
Sameeh

KodyWiremane, 2019-11-14:
@sameehj , contacted. Same title as this issue.

KodyWiremane, 2019-11-19:
Investigated: in Debian 10, ena.ko is shipped with the kernel; currently reported version 2.0.2K. Same with the other driver. The contents of DKMS must be an artifact of the original installation (Debian 8, iirc). I removed the outdated sources, no messing with DKMS now.

A big thanks to @sameehj for a quick response), and sorry for opening an irrelevant issue.

closed 2019-11-19 by KodyWiremane