amzn/amzn-drivers #119: Cross instance IPIP traffic stops working with 2.2.0

willthames, 2020-01-29:
We run EKS on AWS, using AWS’ Amazon Linux 2 base AMI with a custom 5.3.1 kernel on top, built with the ENA driver. I recently changed the build process slightly, and rebuilt the AMI and that stopped cross-node TCP traffic.

After a number of rebuilds to bisect the breakage, I realised the newest AMI was being built with the latest ENA driver (2.2.1 at the time) - I tested identical AMI builds other than ENA versions 2.1.4 and 2.2.0 to prove that the problem starts with 2.2.0

I haven’t yet been able to isolate what particular difference in 2.2.0 is causing this problem. I also don't really understand what specifically is failing! 

We’re running calico as our network fabric for Kubernetes, which encapsulates traffic over IP-in-IP. Traffic to pods on other nodes should be routed out the appropriate tunnel interface and then routed to the appropriate pod on the other side.

I can see from tcpdump on the source host (with 2.2.0 installed) that the SYN packet is sent, but it is not received on the target host (with 2.1.4 installed). 

The SYN is not received:
* if source has 2.2.0 installed
* if target has 2.2.0 installed

So cross-node traffic is currently only working between nodes with 2.1.4 (or lower) installed - and that is cross-node and same AZ or cross-AZ.

Working tcpdump from host layer

```
06:17:10.031639 IP 172.20.12.63 > 172.20.12.36: IP 100.64.21.128.49546 > 100.105.84.93.8080: Flags [S], seq 535392982, win 65535, options [mss 1400,sackOK,TS val 1604145774 ecr 0,nop,wscale 9], length 0 (ipip-proto-4)
06:17:10.031759 IP 172.20.12.36 > 172.20.12.63: IP 100.105.84.93.8080 > 100.64.21.128.49546: Flags [S.], seq 1757528860, ack 535392983, win 65535, options [mss 1400,sackOK,TS val 796287306 ecr 1604145774,nop,wscale 9], length 0 (ipip-proto-4)
06:17:10.031810 IP 172.20.12.63 > 172.20.12.36: IP 100.64.21.128.49546 > 100.105.84.93.8080: Flags [.], ack 1, win 128, options [nop,nop,TS val 1604145774 ecr 796287306], length 0 (ipip-proto-4)
06:17:10.031837 IP 172.20.12.63 > 172.20.12.36: IP 100.64.21.128.49546 > 100.105.84.93.8080: Flags [P.], seq 1:83, ack 1, win 128, options [nop,nop,TS val 1604145774 ecr 796287306], length 82: HTTP: GET / HTTP/1.1 (ipip-proto-4)
06:17:10.031897 IP 172.20.12.36 > 172.20.12.63: IP 100.105.84.93.8080 > 100.64.21.128.49546: Flags [.], ack 83, win 128, options [nop,nop,TS val 796287306 ecr 1604145774], length 0 (ipip-proto-4)
06:17:10.031986 IP 172.20.12.36 > 172.20.12.63: IP 100.105.84.93.8080 > 100.64.21.128.49546: Flags [P.], seq 1:348, ack 83, win 128, options [nop,nop,TS val 796287306 ecr 1604145774], length 347: HTTP: HTTP/1.1 302 Moved Temporarily (ipip-proto-4)
06:17:10.032004 IP 172.20.12.63 > 172.20.12.36: IP 100.64.21.128.49546 > 100.105.84.93.8080: Flags [.], ack 348, win 131, options [nop,nop,TS val 1604145775 ecr 796287306], length 0 (ipip-proto-4)
06:17:10.032096 IP 172.20.12.63 > 172.20.12.36: IP 100.64.21.128.49546 > 100.105.84.93.8080: Flags [F.], seq 83, ack 348, win 131, options [nop,nop,TS val 1604145775 ecr 796287306], length 0 (ipip-proto-4)
06:17:10.032262 IP 172.20.12.36 > 172.20.12.63: IP 100.105.84.93.8080 > 100.64.21.128.49546: Flags [F.], seq 348, ack 84, win 128, options [nop,nop,TS val 796287306 ecr 1604145775], length 0 (ipip-proto-4)
06:17:10.032276 IP 172.20.12.63 > 172.20.12.36: IP 100.64.21.128.49546 > 100.105.84.93.8080: Flags [.], ack 349, win 131, options [nop,nop,TS val 1604145775 ecr 796287306], length 0 (ipip-proto-4)
```

Not working
```
06:20:10.065102 IP 172.20.12.63 > 172.20.13.97: IP 100.64.21.128.32948 > 100.64.187.73.8080: Flags [S], seq 2393751571, win 65535, options [mss 1400,sackOK,TS val 994651410 ecr 0,nop,wscale 9], length 0 (ipip-proto-4)
06:20:11.075544 IP 172.20.12.63 > 172.20.13.97: IP 100.64.21.128.32948 > 100.64.187.73.8080: Flags [S], seq 2393751571, win 65535, options [mss 1400,sackOK,TS val 994652420 ecr 0,nop,wscale 9], length 0 (ipip-proto-4)
06:20:11.843563 IP 172.20.12.63 > 172.20.13.97: IP 100.64.21.129.35064 > 100.64.187.65.3080: Flags [S], seq 4071914450, win 65535, options [mss 1400,sackOK,TS val 1431756228 ecr 0,nop,wscale 9], length 0 (ipip-proto-4)
06:20:13.095548 IP 172.20.12.63 > 172.20.13.97: IP 100.64.21.128.32948 > 100.64.187.73.8080: Flags [S], seq 2393751571, win 65535, options [mss 1400,sackOK,TS val 994654440 ecr 0,nop,wscale 9], length 0 (ipip-proto-4)
```

Pings are returned fine from pods on the target host so this may be isolated to a TCP problem only.

Please let me know if there are any diagnostic tools I can run to further isolate this problem.

akiyano, 2020-01-29:
Hi @willthames,

Thanks for the report. 
We are looking into this.
We will update you once we have (or need) more information.

Thanks,
Arthur

akiyano, 2020-01-29:
Hi @willthames,

We were able to reproduce your issue using the basic ipip tunnel setup described here:
https://sites.google.com/site/mrxpalmeiras/linux/create-ipip-tunnel-between-networks

We are now working on finding the root cause and a fix for it. Once we have one we will update you.

Thanks,
Arthur

akiyano, 2020-01-30:
Hi @willthames,
We found the commit that causes this issue.
You now have 2 options as a workaround until we fix the issue, choose the option more convenient for you:
1. Turn off tx offloads via ethtool using the following command (change ens5 to the name of your interface): 
sudo ethtool -K ens5 tx off

OR

2. Apply the attached patch to driver 2.2.2g using the following command: 
git am 0001-Revert-revert-VXLAN-tx-csum-offloading-support.patch
[0001-Revert-revert-VXLAN-tx-csum-offloading-support.zip](https://github.com/amzn/amzn-drivers/files/4133019/0001-Revert-revert-VXLAN-tx-csum-offloading-support.zip)

Once we have a fix to this issue we will release it here.

Thanks,
Arthur

willthames, 2020-01-31:
@akiyano my workaround is to pin my driver to 2.1.4 for now :) But thanks for the updates!

akiyano, 2020-02-06:
Hi @willthames,

Small update - We've released version 2.2.3 that reverted the commit that gave you this issue.
So you can use it until we release an official fix in a future release.

Regards,
Arthur

akiyano, 2020-02-27:
Closing this since it works.
Sometime in the future we will release a version where it will work with csum offloading.

closed 2020-02-27 by akiyano