amzn/amzn-drivers #120: Update the release note for ENA OS support

yogx4u, 2020-01-30:
I successfully built ENA drives for RHEL 6.6 which is not specified in the release notes. But i couldn't build it for RHEL 6.5

Also Centos6/7 is specified in the Release notes. Specific versions would be great.

RHEL 7.0 gave compilation errors. Is there a plan to support it?

What about Suse SLES 15?

akiyano, 2020-01-30:
Hi @yogx4u,

We will update the release notes in the next release of the driver.
For now, whatever is currently in the release notes is guaranteed to work and other versions are not (although many of them probably do work).
Specifically, I just compiled the driver in RHEL 7.0 (which is currently not written in the release notes as supported) using public ami RHEL-7.0_HVM_GA-20150209-x86_64-1-Hourly2-GP2 (ami-60a1e808) in North Virgina, and it compiled without any errors.
SLES 15 is supported.

Regards,
Arthur

yogx4u, 2020-02-03:
Thank you @akiyano

yogx4u, 2020-02-17:
hi @akiyano 

I tried building ena KO for RHEL 7.0 with uname -r value 3.10.0-123.el7.x86_64.

It failed with error 

> error: implicit declaration of function ‘ether_addr_copy’ [-Werror=implicit-function-declaration]

I did not check the AMI you mentioned but i cannot build the dirvers on my machine.

akiyano, 2020-02-17:
Hi @yogx4u,

Please contact me via akiyano@amazon.com and we will try to solve your specific issue.

Thanks,
Arthur

akiyano, 2022-06-26:
Release notes have been updated with the release of driver 2.7.3g. Closing this issue.

closed 2022-06-26 by akiyano