amzn/amzn-drivers #121: Compile fails on RHEL 6.5

shabin-slr, 2020-02-08:
Make is failing with below errors

`make[1]: Entering directory `/usr/src/kernels/2.6.32-431.29.2.el6.x86_64'
  CC [M]  /tmp/amzn-drivers-master/kernel/linux/ena/ena_netdev.o
In file included from /tmp/amzn-drivers-master/kernel/linux/ena/ena_netdev.h:36,
                 from /tmp/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:52:
/tmp/amzn-drivers-master/kernel/linux/ena/kcompat.h:233:27: error: net/busy_poll.h: No such file or directory
/tmp/amzn-drivers-master/kernel/linux/ena/ena_netdev.c: In function ‘ena_set_rx_hash’:
/tmp/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:1623: error: storage size of ‘hash_type’ isn’t known
compilation terminated due to -Wfatal-errors.
make[2]: *** [/tmp/amzn-drivers-master/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/tmp/amzn-drivers-master/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/2.6.32-431.29.2.el6.x86_64'
make: *** [all] Error 2
`

shabin-slr, 2020-02-08:
AMI ID : ami-00a11e68

akiyano, 2020-02-09:
Hi @shabin-slr,

This version is not officially supported by ENA.
The earliest supported RHEL version is 6.6.

Regards,
Arthur

closed 2020-02-09 by akiyano