amzn/amzn-drivers #123: Support for ethtool --set-channels

eugeneia, 2020-02-26:
As far as I know it is currently only possible to adjust the number of channels for *all* ENA interfaces via the `num_io_queues` module parameter.

Is support for `ethtool --set-channels` (i.e., per interface number of rx/tx queue configuration) planned?

akiyano, 2020-02-26:
Hi @eugeneia,

Actually it is already supported :)

Example using the latest driver version 2.2.3:

```
[ec2-user@ip-172-31-12-68 ena]$ ethtool -l eth0
Channel parameters for eth0:
Pre-set maximums:
RX:		0
TX:		0
Other:		0
Combined:	2
Current hardware settings:
RX:		0
TX:		0
Other:		0
Combined:	2
[ec2-user@ip-172-31-12-68 ena]$ sudo ethtool -L eth0 combined 1
[ec2-user@ip-172-31-12-68 ena]$ ethtool -l eth0
Channel parameters for eth0:
Pre-set maximums:
RX:		0
TX:		0
Other:		0
Combined:	2
Current hardware settings:
RX:		0
TX:		0
Other:		0
Combined:	1

[ec2-user@ip-172-31-12-68 ena]$ ethtool -l eth1
Channel parameters for eth1:
Pre-set maximums:
RX:		0
TX:		0
Other:		0
Combined:	2
Current hardware settings:
RX:		0
TX:		0
Other:		0
Combined:	2
```
Enjoy,
Arthur

closed 2020-02-26 by akiyano

eugeneia, 2020-02-27:
@akiyano Can you recommend a minimum kernel version for use with ena 2.2.3? I am testing with a 5.4.22 kernel, and ena 2.2.3 seems to break XDP on that kernel at least (the stock ena that comes with 5.4.22 worked).

eugeneia, 2020-02-28:
Reply to self: the 5.5 series seems to be fully functional and includes the ena 2.2.3! 🎉️

sameehj, 2020-03-01:
Hi @eugeneia ,

I have tried to reproduce the "break" of XDP in 5.4.22 along with ena 2.2.3 but without success. I did the following steps:

* Used Amazon Linux 2 AMI- ami-0a887e401f7654935 in northern Virginia on top of which I have installed Linux 5.4.22 from the stable repository (https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/)
* Cloned amzn-drivers and managed to compile the ena driver with and without the "ENA_XDP_SUPPORT" define.
* Managed to run a XDP sample program from the kernel samples.

Can you please elaborate on what issues did you face that broke XDP for you?

You can contact me on sameehj@amazon.com to make this easier :)

Thanks,
Sameeh

kcn21, 2021-10-21:
Hey @akiyano @eugeneia , I got the error "Cannot set device channel parameters: Invalid argument"
when I use sudo ethtool -L enp4s0 combined 4 command.

My output of ethtool -l enp4s0 command is following:

Channel parameters for enp4s0:
Pre-set maximums:
RX:		4
TX:		1
Other:		0
Combined:	0
Current hardware settings:
RX:		4
TX:		1
Other:		0
Combined:	0

Please help me regarding this issue.

techthumb, 2022-04-26:
Is there a way to explicitly set the `tx` and `rx` values?

```
$ ethtool --set-channels eth1 rx 4
Cannot set device channel parameters: Invalid argument
```

```
$ cat /etc/system-release
Amazon Linux release 2 (Karoo)

$ ethtool --show-channels eth1
Channel parameters for eth1:
Pre-set maximums:
RX:		0
TX:		0
Other:		0
Combined:	8
Current hardware settings:
RX:		0
TX:		0
Other:		0
Combined:	4

$ ethtool --set-channels eth1 combined 8

$ ethtool --show-channels eth1
Channel parameters for eth1:
Pre-set maximums:
RX:		0
TX:		0
Other:		0
Combined:	8
Current hardware settings:
RX:		0
TX:		0
Other:		0
Combined:	8

$ ethtool --set-channels eth1 combined 4

$ ethtool --set-channels eth1 rx 4
Cannot set device channel parameters: Invalid argument

$ uname -a
Linux ip-10-21-120-92.ec2.internal 5.10.109-104.500.amzn2.x86_64 #1 SMP Wed Apr 13 20:31:43 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

akiyano, 2022-04-27:
Hi @techthumb ,
No there isn't a way to set rx or tx values separately.
When you set the combined value it sets both rx and tx values to the same value.

ShayAgros, 2022-04-27:
@techthumb out of curiosity, what's the motivation to set TX and RX separately ?

Would such capability improve a certain flow ?

techthumb, 2022-04-27:
@akiyano I am attempting to use https://github.com/Xilinx-CNS/onload/ with ENA.

When registering the NIC with onload, I was running into this error:

```
[  443.536158] ena 0000:00:06.0 ens6: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (8), the maximal queue count (8)
```

I ended up setting the `combined` value to `4` which satisfied this condition.