amzn/amzn-drivers #124: multi-threading model with EFA

zarzen, 2020-03-01:
Hi there,
I am trying to use EFA via `libfabric` interface. and I want to use multi-threading to saturate 100Gbps bandwidth. 
The reason for multi-threading is because the `fi_rdm_tagged_bw` test on two `c5n.18xl` instances can achieve roughly 50% bandwidth utilization. Thus, I was thinking to launch two threads for data transfering.
```
bytes   iters   total       time     MB/sec    usec/xfer   Mxfers/sec
64      20k     1.2m        0.03s     43.77       1.46       0.68
256     20k     4.8m        0.03s    165.03       1.55       0.64
1k      20k     19m         0.03s    605.67       1.69       0.59
4k      20k     78m         0.04s   2313.86       1.77       0.56
64k     2k      125m        0.02s   5851.43      11.20       0.09
1m      200     200m        0.03s   6734.38     155.71       0.01
```

But I am not sure what is a good to use multi-threading with `EFA`. I have two options in mind:
* 1) single `fi_endpoint` with multi-threads send/recv data over single `fi_endpoint`
* 2) multiple threads, each thread has its own `fi_endpoint` for send/recv data

For the second option, I am suspecting the viability. Because I am not sure if it is possible to have two or more endpoints associated with `EFA`. (When I launched `fi_rdm_tagged_bw` server with `-B 9999` option and connecting the launched server with `-P 9999` option, two processes cannot be connected. Full commands show following)

```
# server
fi_rdm_tagged_bw -p efa -s fe80::3f:6fff:fed1:f7ff -B 9999

# client, -P 9999 not work
fi_rdm_tagged_bw -p efa -P 9999 -s fe80::3f:6fff:fed1:f7ff  fe80::4d:e8ff:feb1:71ef

# client, -P 0 flag works 
fi_rdm_tagged_bw -p efa -P 9999 -s fe80::3f:6fff:fed1:f7ff  fe80::4d:e8ff:feb1:71ef
```

firasj, 2020-03-03:
Hello @zarzen,
I can see that you've opened another issue on ofiwg/libfabric.
Do you require any further assistance here?

zarzen, 2020-03-03:
Hi @firasj 
Wei replied me at that thread, while we didn't talk specifically on the threading model there. 
At that discussion, we mainly talked about the rationality behind the numbers I measured. 
At the same time he did mention that is possible to have multiple EP work with EFA, and pointed me to OSU-MPI-Benchmark(MVAPICH). 
While it is still not clear to me how to start with multiple endpoints or multi-threading with EFA/libfabric, if you could give certain suggestions that would be great, thanks!

firasj, 2020-03-05:
Hi @zarzen ,
This discussion might be better asked over libfabric because it's not related to the EFA driver.
However, looking at your command, I suggest to use OOB address exchange which will save you some trouble because EFA does not use the conventional source/destination port numbers.

Regardless, EFA does support running multiple endpoints, associated with same PD or not.
For example you can run:
```
# server
fi_rdm_tagged_bw -p efa -E=3000 &
fi_rdm_tagged_bw -p efa -E=3001 &
# client
fi_rdm_tagged_bw -p efa -E=3000 <server ip> &
fi_rdm_tagged_bw -p efa -E=3001 <server ip> &
```

and it will run 2 applications in parallel.

zarzen, 2020-03-05:
> fi_rdm_tagged_bw -p efa -E=3000

Hi @firasj, thanks for your suggestion, 
but I don't think those two endpoints can communicate through IP. 
And `-E` is not a valid argument for `fi_rdm_tagged_bw` I think.
```
ubuntu@ip-172-31-12-1:~$ fi_rdm_tagged_bw -p efa -E 3000
connect: Invalid argument
```
If using `fi_rdm_tagged_bw -p efa -E=3000` then, ` -E=3000` probably interpreted as destination address.

firasj, 2020-03-05:
Looking at the source code: https://github.com/ofiwg/libfabric/blob/master/fabtests/benchmarks/rdm_tagged_bw.c#L80, the test has `CS_OPTS` which includes the `-E` option.
Are you sure you're using latest libfabric/fabtests?

zarzen, 2020-03-05:
yes, I am running fabtests-1.9.0
Even though the `-E` can work, the program still fail with IP:
```
# server
$ ./fi_rdm_tagged_bw -p efa -E=3000ft_fw_recv(): /home/ubuntu/fabric-test/src/common/shared.c:3172, ret=-1 (Operation not permitted)
recv: Connection reset by peer

# client
$ ./fi_rdm_tagged_bw -p efa -E=3000 127.0.0.1
fi_getinfo(): /home/ubuntu/fabric-test/src/common/shared.c:729, ret=-61 (No data available)
```

zarzen, 2020-03-06:
closing due to https://github.com/ofiwg/libfabric/issues/5696

closed 2020-03-06 by zarzen