amzn/amzn-drivers #125: ENA kernel driver keeps on printing message "unregister_netdevice: waiting for eth1" on linux kernel 4.19.51

hcalcuttawala-a10, 2020-04-21:
Hi,

I am trying to boot an AWS instance of instance type (m4 16xl) on AWS cloud with 2 ENA supported interfaces. I have observed that as the bootup completes, and when I try to remove one of the interface from Kernel Driver to bind to igb_uio driver, the varlog/messages gets filled with error as shown below,

```
-bash-4.2# uname -r
4.19.51

-bash-4.2# ethtool -i eth0
driver: ena
version: 1.5.0K
firmware-version:
expansion-rom-version:
bus-info: 0000:00:03.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

[12620.189810] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12630.301701] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12640.381721] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12650.461718] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12660.541712] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12670.621701] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12680.701699] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12690.781713] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12700.861703] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12710.941710] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12721.021697] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12731.101704] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12741.181697] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12751.261792] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12761.341700] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12771.421718] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12781.501699] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12791.581700] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12801.661728] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12811.741720] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12821.821696] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12831.933696] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12842.013704] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12852.093741] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12862.173716] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12872.253796] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12882.333875] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12892.413718] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
[12902.525730] unregister_netdevice: waiting for eth1 to become free. Usage count = -1
```

Can you please let me know if this was observed with ENA kernel driver and whether its been addressed in any release.

Thanks.

zorikm, 2020-04-21:
Hi,
Looks like you are using an old driver version. We strongly suggest to update your ENA driver.

Thanks

sameehj, 2020-04-22:
Hi @hcalcuttawala-a10,

This is a known issue, you are using an old kernel driver. The issue was resolved in github driver >=2.1.0g (we recommend using the latest 2.2.7g)  and in kernel 4.20 or above. You can use the GitHub driver or else upgrade the kernel.

Thanks for reporting this,
Sameeh

hcalcuttawala-a10, 2020-04-22:
Hi Sameeh,

Thanks, can you please let me know the recommended version of ena driver from github for 4.19.51 kernel version.

Thanks

sameehj, 2020-04-22:
@hcalcuttawala-a10 the latest version (currently 2.2.7g) is the recommended one.

hcalcuttawala-a10, 2020-04-23:
@sameehj, Thanks, can you please share the commit which resolves this issue with kernel 4.19.51.

zorikm, 2020-04-23:
@hcalcuttawala-a10,
We strongly suggest to switch to the latest driver version. You use an outdated driver, and if you don't upgrade you might face additional issues.

hcalcuttawala-a10, 2020-04-23:
Thanks @zorikm and @sameehj for your replies, will upgrade to latest driver. Closing this issue.

closed 2020-04-23 by hcalcuttawala-a10