amzn/amzn-drivers #126: Send generated traffic (tcpreplay, t-rex) between two ena devices

norg, 2020-04-28:
I have a EC2 instance with 3 interfaces (ena) while the first is used for management (ens5) I have the 2nd and 3rd interface (ens6/ens7) that I want to use to generate traffic between those two. I want to use T-rex (which uses dpdk) and/or tcpreplay to send crafted traffic between ens6 and ens7. I also created a catch all traffic mirror filter for that since I want to capture that traffic on another ec2 instance.

If I send valid traffic on those interfaces, like ICMP it is seen on the capture device. But the traffic generated by t-rex and tcpreplay is not seen on the traffic mirror target. But t-rex and tcpreplay work as expected since I can see the traffic on ens6 and ens7 when I look into tcpdump there. I even tried to craft traffic with the IPs configured on those interfaces but that didn't help either.

So is there a magic happen with ena or within AWS that doesn't allow such generated traffic to be mirrored while all other "real" traffic works?

I made sure that the filter is a catchall and that the securitygroup is liberal by allowing the traffic.

nafeabshara, 2020-04-28:
Andi,

Could you share sample packets being sent?

Packet drop could happen when malformed headers, bad checksum flags, or wrong source Mac being used in a given interface 

Could u also share Ena dpdk driver version ?

zorikm, 2020-04-28:
Andi,
In addition to the above it would be helpful if you could provide instance-id of your Mirroring Source and Target instances.
Thanks.

norg, 2020-04-29:
I looked again into the pcaps and was able to enforce to use the correct MACs and it worked at least with the t-rex run that uses the linux interfaces. I will retry with the high performance one where t-rex uses dpdk to aquire the whole interface.

I use t-rex 2.79 (2.80 wasn't able to run). Is there an easy way to figure out what ena dpdk driver version is used there?

If I see any issues with the high-performance testrun I will update the issue and also provide the instance-ids.

Thanks for the hint with possible reasons for the drops already!

norg, 2020-04-29:
The DPDK enhanced mode runs as well, thanks a lot for your help!

closed 2020-04-29 by norg

I-gor-C, 2020-04-29:
> 
> I use t-rex 2.79 (2.80 wasn't able to run). Is there an easy way to figure out what ena dpdk driver version is used there?
> 
Per [T-Rex 2.79 repo](https://github.com/cisco-system-traffic-generator/trex-core/blob/fe76d22b60c00c4e6a0ed6897bc2395c47753d55/src/dpdk/drivers/net/ena/ena_ethdev.c), it includes ENA driver 2.0.0. In general, the ENA driver and DPDK versions matching can be found in the official Amazon drivers repository: https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/RELEASENOTES.md