amzn/amzn-drivers #128: XDP_REDIRECT Support

doonhammer, 2020-05-01:
Is there a timeline for XDP_REDIRECT support ?

gtzalik, 2020-05-05:
Hi @doonhammer,
We are currently working on adding XDP_REDIRECT to ENA driver. 
I can't share the timeline at this point but I will update here on our progress.

doonhammer, 2020-05-05:
Thanks @gtzalik appreciate the update.

talawahtech, 2020-07-11:
@gtzalik my understanding is that AF_XDP (one of the main targets of XDP_REDIRECT) support didn't make it into the Linux kernel until 4.18 or 4.19. Do you know if there are any plans to backport AF_XDP support to the 4.14 kernel used by Amazon Linux 2?

akiyano, 2020-08-27:
@talawahtech There are currently no plans to backport AF_XDP support to the 4.14 AL2 kernel.

washley, 2020-08-27:
Why was this closed? The backport question was not the issue, adding XDP_REDIRECT support is.

akiyano, 2020-08-27:
@washley you are right, I'll leave it open for tracking and will close it once XDP_REDIRECT is released

akiyano, 2020-08-27:
XDP_REDIRECT is currently expected to be released in Q4 of 2020

doonhammer, 2020-08-27:
@akiyano thanks for the update.

Regards

John

maryamtahhan, 2020-11-19:
Hey Folks

just wondering if this is still planned for Q4 2020? and in that case will it support redirect to AF_XDP sockets?

Thanks

akiyano, 2020-11-19:
Hi @maryamtahhan,

XDP_REDIRECT will be in the next version we expect to release in the next few weeks.
It will also be submitted to the upstream Linux kernel at around the same time.

AF_XDP sockets support is planned for a future version.
I can't share the timeline at this point but I will update here on our progress.

Regards,
Arthur

vcunat, 2021-01-22:
It's in 2.3.0 release notes: https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/RELEASENOTES.md#r230-release-notes

akiyano, 2021-01-22:
@vcunat Yes you are correct XDP_REDIRECT was released as part of version 2.3.0

closed 2021-01-22 by akiyano