amzn/amzn-drivers #130: Any plans to distribute updated ena driver for Amazon Linux 2

talawahtech, 2020-05-26:
Are there any plans to package/distribute more recent versions of the ena driver for Amazon Linux 2? 2.1.3 appears to be the latest version available with AL2 AMIs.

I am troubleshooting what appears to be an RSS related issue that I am hoping is resolved in a more recent version of the driver.

I know it is possible to build from source, but that is less than ideal when automatically deploying a cluster of servers replaceable. Any plans to distribute updates via a channel like amazon-linux-extras?

sameehj, 2020-05-26:
Hi @talawahtech,

Thank you for reaching out.  The latest ena-driver should be included in the next Amazon Linux 2 release within the next few weeks. Regarding your RSS issue, I'd be happy to help you to investigate it. To make this efficient, please contact me at sameehj@amazon.com.

Best regards,
Sameeh

akiyano, 2020-07-30:
Hi @talawahtech,

The latest Amazon Linux 2 ami now includes driver 2.2.10g.
(for example see "amzn2-ami-hvm-2.0.20200722.0-x86_64-gp2 (ami-02354e95b39ca8dec)" in n. virgina), 

Best regards,
Arthur

closed 2020-07-30 by akiyano

talawahtech, 2020-08-02:
Awesome, thanks. I am using that version now.

FYI, what I originally thought was an RSS issue actually turned out to be an issue with how SO_ATTACH_REUSEPORT_CBPF works with forked processes that are pinned to a specific CPU.

SO_ATTACH_REUSEPORT_CBPF works best if the order in which the sockets are opened matches the index of the CPU to which the process is pinned. I had a for loop that was forking a child process, pinning it to a CPU and then starting up my http server (which then opens the socket to listen on). However since fork is being called in a loop in the parent, the order in which the sockets actually got opened in the child processes was not deterministic.

I ended up using an eventfd object to communicate between the parent and child processes and ensure that the forking happened sequentially so that the order of the sockets being opened matched the order of the CPUs being pinned.

Just figured I should share in case anyone else runs into the same issue.