amzn/amzn-drivers #135: ENA DPDK  RSS key,hash capability

hhaim, 2020-07-28:
DPDK 20.02 but even on latest the code does not exist 

1. Ignore RSS key  the key/len from 

	```dev->data->dev_conf.rx_adv_conf.rss_conf.rss_key;
	   dev->data->dev_conf.rx_adv_conf.rss_conf.rss_key_len
        ```
2. There is no way to change the hash to toepoiltz 

```
    if (rte_eth_dev_filter_supported(m_repid, RTE_ETH_FILTER_HASH)) {  << should return support (zero)
        // Setup HW to use the TOEPLITZ hash function as an RSS hash function
        struct rte_eth_hash_filter_info info = {};
        info.info_type = RTE_ETH_HASH_FILTER_GLOBAL_CONFIG;
        info.info.global_conf.hash_func = RTE_ETH_HASH_FUNCTION_TOEPLITZ;
        if (rte_eth_dev_filter_ctrl(m_repid, RTE_ETH_FILTER_HASH,
                                    RTE_ETH_FILTER_SET, &info) < 0) {
        }
    }

```
related issue is:

https://github.com/cisco-system-traffic-generator/trex-core/issues/509

semihalf-krawczyk-michal, 2020-07-30:
Hi Hanoh,

the RSS hash key set setup isn't mandatory feature of the PMD. In the table [over there](https://doc.dpdk.org/guides/nics/overview.html) you can see all features that ENA PMD supports (this one is called `RSS key update`).

Anyway, the current hardware doesn't support setting custom RSS hash function or the hash key - we're setting the CRC32 hash, but if you investigate the result closer, you would see, that if returns `EOPNOTSUP`. It'll be the same result when you would try to change HF to Topetlitz or set the custom key.

Effectively the device will be using it's default RSS key and HF (which should be Topetlitz) and it doesn't matter what the PMD does on that field.

The PMD support for setting hash key or function will be added after those features will be available in the HW.

Thanks,
Michal

hhaim, 2020-07-30:
@semihalf-krawczyk-michal, thanks for the clarification. Yes, I did dig into this and saw that the hardware ignore the RSS_HASH type and key. When you say "available in the HW", is it a firmware update (software) or a new ASIC which will take longer to support? 
In any case we will need to find a software redirect solution (that will incur a performance penalty) for this types of NICS that does not support RSS KEY change.

closed 2020-07-30 by hhaim

semihalf-krawczyk-michal, 2020-07-30:
@hhaim I'm sorry for the inaccurate information - I've just received information that the RSS hash key/function reconfiguration is already supported on the latest generation of the instances (c5n/r5n) - but it requires extra actions from the driver perspective.

We're working on a patch that's going to add RSS hash key/function reconfiguration support and we're planning to release it in the DPDK v20.11.

Thanks,
Michal

hhaim, 2020-07-30:
@semihalf-krawczyk-michal thanks for the update. Would be nice to get an early patch ( I don't see it on the latest). 
I've tested my patch on c5n and got "unsupported" from the hardware, so I probably miss something.

priikone, 2020-12-03:
What's the status of this support in ena?  It doesn't look like it made it to DPDK 20.11.

This bug probably shouldn't have been closed...

semihalf-krawczyk-michal, 2020-12-04:
Hi @hhaim and @priikone ,

we postponed release of this feature to the next DPDK version. Currently the patch is during internal review process.

If you're still interested in testing the patch internally (it'll be based on v20.11 DPDK), please let us know and we'll provide you with the appropriate code.

Thanks,
Michal

priikone, 2020-12-04:
Yes, please.

semihalf-krawczyk-michal, 2020-12-04:
We'll share it as soon as the internal review will be completed.