amzn/amzn-drivers #136: RSS Hash Key and toeplitz hashing on ENA

suddas, 2020-07-30:
Can any one paste exact RSS Hash implementation (toeplitz  source code), hash key, and hash input that is implemented in NIC.

suddas, 2020-07-31:
Using following 32b toeplitz. 
driver version version: 2.2.10g. 
It seems nic using 16b implementation. How to make this 32 bits implementation into 16 bits?
  
uint32_t
  compute_softrss(const uint32_t *input_tuple, uint32_t input_len, const uint8_t *rss_key)
  {
      uint32_t i, j, map, ret = 0;

      for (j = 0; j < input_len; j++) {
          for (map = input_tuple[j]; map; map &= (map - 1)) {
                  i = (uint32_t)__builtin_ctz(map);
                  ret ^= cpu_to_be32(((const uint32_t *)rss_key)[j]) << (31 - i) |
                              (uint32_t)((uint64_t)(cpu_to_be32(((const uint32_t *)rss_key)[j + 1])) >>
                              (i + 1));
          }
      }
      return ret;
  }

akiyano, 2020-07-31:
Hi @suddas ,

Our Toeplitz implementation for c5n type instances should match a standard Toeplitz implementation.

Using a c5n instance with Amazon Linux 2 that has driver 2.2.10g installed,
then adding the following print:
printk("%s(%d): hash is %08x\n", __func__, __LINE__, ena_rx_ctx->hash);
to ena_set_rx_hash() you should see a 32 bit hash.

Please contact akiyano@amazon.com for further assistance.

Thanks,
Arthur

suddas, 2020-08-20:
Thanks you so much @akiyano. RSS is working for me now.

closed 2020-08-20 by suddas