amzn/amzn-drivers #137: Issue with tc matches using ENA instances

jfoster-twilio, 2020-07-31:
# Background
I've been working with tc filters to match ingress traffic based on UDP destination port(s). In testing, I noticed that when sending UDP payloads in excess of 214 bytes, the filter would not match the traffic towards the target UDP destination port(s). I have observed this behavior on AWS instances with ENA (e.g. c5d, t3) but was not able to reproduce on older generation instances (e.g. t2) which I believe use a different network driver. I believe this may be due to an issue with the ENA driver.

For testing I have setup the following an ingress qdisc with the following filter:
```bash
sudo tc qdisc add dev eth0 handle ffff: ingress
sudo tc filter add dev eth0 parent ffff: protocol ip pref 1 basic match "cmp(u16 at 2 layer transport eq 12400)" action simple sdata "dport_12400"
```

The expectation is that the 2 byte value at offset 2 from the transport header will have value equal to 12400 (e.g. the destination port field equals 12400), and the message `dport_12400` will be printed for matches

The following describes the testing setup I've used for comparison which can be used to reproduce the issue:

# ENA instance
Launch a t3.large instance with Amazon Linux 2 Image (AMI ID amzn2-ami-hvm-2.0.20200722.0-x86_64-gp2 (ami-0873b46c45c11058d))

Install tc and setup the ingress qdisc and filter:
```bash
sudo yum install -y tc
sudo tc qdisc add dev eth0 handle ffff: ingress
sudo tc filter add dev eth0 parent ffff: protocol ip pref 1 basic match "cmp(u16 at 2 layer transport eq 12400)" action simple sdata "dport_12400"
```

Send a UDP packet to (instance public hostname, dport 12400) with payload size 200 bytes

I've used the following python script in my testing
```python
import socket

UDP_IP = "ec2-44-234-39-17.us-west-2.compute.amazonaws.com"
UDP_PORT = 12400

print "UDP target IP:", UDP_IP
print "UDP target port:", UDP_PORT

len = int(input("Please enter a payload length: "))
MESSAGE = "A" * len
print "message:", MESSAGE

sock = socket.socket(socket.AF_INET, # Internet
             socket.SOCK_DGRAM) # UDP
sock.bind(('0.0.0.0', 50020))
sock.sendto(MESSAGE, (UDP_IP, UDP_PORT))
```

Now you should see a match for this packet printed, e.g.:
```bash
dmesg | grep dport_12400
[  100.588139] simple: dport_12400_1
```

Next, send another UDP packet to (instance public hostname, dport 12400) with payload size 400 bytes

Check and see, there was no match printed for the second packet. e.g.
```bash
dmesg | grep dport_12400
[  100.588139] simple: dport_12400_1
```

In my probing, I've found that UDP payloads with length in excess of 214 bytes do not match this filter

# Non-ENA Instance
Now, if you instead launch a t2.large instance with Amazon Linux 2 Image (amzn2-ami-hvm-2.0.20200722.0-x86_64-gp2 (ami-0873b46c45c11058d)), you will find that when running the same tests, UDP packets match the filter as expected (including those whose payload size exceeds 214 bytes).

# Instrumentation
I've done a bit more debugging to try and confirm where the issue lies by instrumenting the kernel module which is doing the matching here (net/sched/em_cmp.c) and found that the following section is where the matching fails for larger packets:
https://github.com/torvalds/linux/blob/master/net/sched/em_cmp.c#L28

The relevant check does the following (https://github.com/torvalds/linux/blob/master/include/net/pkt_cls.h#L493-L499)

[... 224 lines cut ...]
  .kind	  = TCF_EM_CMP,
  .datalen  = sizeof(struct tcf_em_cmp),
  .match	  = em_cmp_match,
  .owner	  = THIS_MODULE,
  .link	  = LIST_HEAD_INIT(em_cmp_ops.link)
};

static int __init init_em_cmp(void)
{
    //pr_info("em_cmp Loaded\n");
    return tcf_em_register(&em_cmp_ops);
}

static void __exit exit_em_cmp(void)
{
  tcf_em_unregister(&em_cmp_ops);
}

MODULE_LICENSE("GPL");

module_init(init_em_cmp);
module_exit(exit_em_cmp);

MODULE_ALIAS_TCF_EMATCH(TCF_EM_CMP);
```

akiyano, 2020-07-31:
Hi @jfoster-twilio,

Thanks for reporting this issue with such detailed reproduction steps!
What you are describing rings a bell and seems related to a fix to some other issue we are currently working on.
Please check if the attached patch fixes the issue for you and let me know.

To do that please follow these steps:
1. Copy the patch to amzn-drivers directory.
2. Run the following command from inside the amzn-drivers directory to patch the driver:
`git am 0001-copy-pkt-headers-to-skb-inline-data.patch.txt `
3. Build and install the patched driver
4. Run the test that reproduces the issue for you and see if it still happens.

[0001-copy-pkt-headers-to-skb-inline-data.patch.txt](https://github.com/amzn/amzn-drivers/files/5009251/0001-copy-pkt-headers-to-skb-inline-data.patch.txt)

Thanks,
Arthur

jfoster-twilio, 2020-07-31:
Hi @akiyano, thanks for the quick response! My initial tests indicate the patch resolves the reported issue. Would you mind updating this when the patch lands?

akiyano, 2020-08-01:
Hi @jfoster-twilio ,

Great to hear it solves your issue!
I will update this ticket once we release the fix. 
The fix will not be part of the next release, since it is already closed and will be released soon.
But will probably be in the release after that. 

Thanks,
Arthur

jfoster-twilio, 2021-06-02:
@akiyano Any updates on when this will make it into a release?

akiyano, 2021-06-02:
Hi @jfoster-twilio,

Sorry for not updating here.
Please try it out with the latest 2.5.0 driver, which contains the fix to your issue.

Thanks,
Arthur

closed 2021-09-29 by akiyano