amzn/amzn-drivers #138: Performance benchmark - packet per seconds

suddas, 2020-08-11:
Is there any benchmark for packets per second (PPS) - with small (64/1024 byte ) packets for following instances with ena. Running two EC2 instances B2B (one client with some pkt gen, and one server), in same VPC and same placement cluster group.

1. c5n.9xlarge
2. c5n.4xlarge.

AWSNB, 2020-08-11:
@suddas as typical in benchmarks, it is application and workload specific.  so we encourage you to try it out with your specific workloads

last time I checked, if you are running enough flows (certainly need more than single flow to maximize PPS),  you should be able to sustain ~5mpps on c5n.9xl and 2.5mpps on .4xl ,  that was my personal experiment which I thinks others where able to reproduce.   Other factors like whether its TCP vs UDP,  new connection rate/second, and total number of concurrent active connection may results in different performance

is there a specific target you are shooting for or specific issues you think you may be hitting ?

suddas, 2020-08-11:
@AWSNB What was the cpu utilization in your case. My intention is to know if my application performance is limited by IO bottleneck or cpu bottleneck. My application is multi threaded, event driven and scale across the cores.

I am running on c5n.9xl. I am getting bidirectional PPS (Tx  Rate + Rx Rate) = 6mpps, but only using 18 virtual cores. Almost 50 % cpu remain idle.

I am running TCP flow with 1024 active parallel connections -  with one source IP and one destination IP. By enough flows do you mean multiple different source or dest IPs  or multiple parallel connections are sufficient?

Increasing, core count or number of parallel connections does not improve the performance at all in my application

zorikm, 2020-08-12:
@suddas, multiple parallel connections are sufficient.
As to a bottleneck, sounds like you answered your question by yourself. 6mpps seems to be a reasonable packet rate for c5n.9xlarge, and it is not limited by CPUs.

suddas, 2020-08-12:
Thanks a lot @AWSNB  and @zorikm  for your confirmation.

AWSNB, 2020-08-12:
Please email us on nafea@amazon.com and zorik@amazon.com ,  would like to share some unreleased items

closed 2020-08-25 by zorikm