amzn/amzn-drivers #141: ENA driver do not allows mac address to be changed

arun-vinod-rva, 2020-08-28:
Hi,
It is well known that mac address spoofing is not something aws supports for network interfaces and also aware of the fact that aws VPC does not allow traffic from unknown macaddress:IP combination.

However, there are production requirements which requires the mac address of NIC to be changed. consider the case when we are running a L2 / L3 appliance in an ec2 instance and traffic generating from these virtual devices will have source_ip and source_mac generated by virtual switches in these appliances(VPC blocks this traffic). However, the NIC allocated from VPC is configured in a special manner, so that it will not have IP and no traffic will be generated from it, rather it will act more similar to a bridge. So that plan is to reuse this NIC's IP address and mac address to the virtual components, so the packet hitting the NIC will have source mac_address and IP known to VPC and can reach internet. 

For this configuration, the original mac address of the NIC to be changed to some random so that only we can reuse it on the virtual devices inside the ec2 appliance. 

The ENA driver is not allowing for mac_adress change but the same has been success with vif driver. So i request the team to remove this bottleneck from ENA driver so that it will be a relief for those who intend to use ec2 for networking appliances.

zorikm, 2020-08-31:
@arun-vinod-rva,
Thanks for reaching out. 
We would like to understand better some details of your use case.
Could you please reach out to csbisa@amazon.com and zorik@amazon.com.
Thanks!

akiyano, 2020-09-30:
Fix candidate provided to client, waiting for client to test it.

akiyano, 2020-11-23:
patch was provided and if any further help is needed please reopen

closed 2020-11-23 by akiyano

akiyano, 2022-05-22:
Hi, 

Changing the MAC address is not supported by AWS.
A patch can be used to enable changing the MAC address at the driver level, but this renders the eni unusable. 
If this solution is relevant for you, please contact me via akiyano@amazon.com and I’ll provide such a patch.
If this is not good enough, and you have a real use case that requires changing the MAC address, please contact AWS support and we will try to find the best solution for you.