amzn/amzn-drivers #142: Amazon LLQ-enabled interfaces

jha, 2020-09-25:
Hi team,

I wasn't sure what the right avenue to reach out was to find out more about LLQ w/ DPDK.

I've tried to spin up an i3 instance and a z1d instance to check for LLQ support, but the output of lspci -n shows

```
00:05.0 0200: 1d0f:ec20
00:06.0 0200: 1d0f:ec20
```

I can't seem to find much info on it, apart from the commit messages which just allude to "saving a few microseconds"

Thanks in advance!

zorikm, 2020-09-25:
Hi @jha 
LLQs are offered on Nitro hypervisor based instances, to which we switched approximately 2 years ago.
i3 is a relatively old Xen instance, and therefore LLQs are not exposed there.
There are 2 ways for you to figure out if LLQs are supported.
1. Check driver logs in dmesg. The latest driver logs a message if LLQ is not supported.
On i3 for example you would see: LLQ is not supported Fallback to host mode policy.
2. As a rule of thumb, please run lspci -vvv and check if BAR2 (Region 2) is exposed by ENA device. If it is, LLQs are supported.
Z1d for example supports it.
Hope this helps,
Zorik

jha, 2020-09-25:
Hi @zorikm 

Thanks for the fast response on this. I don't see any driver logs in dimes advertising lack of LLQ support on z1d.

lspci -vvv shows region 2 is available.

Based on this, it seems like it's just the documentation that's outdated. See kernel/fbsd/ena/README for the following table

```
Supported PCI vendor ID/device IDs:
===================================
1d0f:0ec2 - ENA PF
1d0f:1ec2 - ENA PF with LLQ support
1d0f:ec20 - ENA VF
1d0f:ec21 - ENA VF with LLQ support
```

The output of lspci -n on my z1d machine is as follows

```
lspci -n
00:00.0 0600: 8086:1237
00:01.0 0601: 8086:7000
00:01.3 0000: 8086:7113 (rev 08)
00:03.0 0300: 1d0f:1111
00:04.0 0108: 1d0f:8061
00:05.0 0200: 1d0f:ec20
00:06.0 0200: 1d0f:ec20
00:1f.0 0108: 1d0f:cd01
```

Slots 5 and 6 are the network adapters, which is why I was concerned that LLQ support was not enabled.

The DPDK documentation is also similarly out of date. Can you confirm if this is the case? Then perhaps I could open an issue to get the docs updated on DPDK as well.

B/R

zorikm, 2020-09-25:
Hi @jha
You are absolutely right about outdated documentation. We'll fix it for fbsd and DPDK asap. Thanks for calling this out.  
As I mentioned in my previous response z1d supports LLQs, i3 doesn't.

Thanks,
Zorik

I-gor-C, 2021-01-11:
The documentation has been updated for both DPDK and FreeBSD.
* DPDK: Please use v20.11 or later
* FreeBSD: Please update to the current driver from this repository

closed 2021-01-11 by I-gor-C