amzn/amzn-drivers #143: VFIO write combining patch doesn't apply cleanly to Ubuntu 18.04

mikeb01, 2020-10-02:
Hi,

I've been testing DPDK in AWS using Ubuntu 18.04 images.  The patch embedded in the `vfio-wc-patch.sh` script does not appear to apply cleanly.  For `linux-aws-5.3-5.3.0` version (initial version for image), it reports and error, but appears to apply the necessary changes.  For `linux-aws-5.4-5.4.0` (version after applying updates) is partially applies and I had to make the rest of the changes manually.  I've posted the diffs below.  If I am wanting to use ENA adapter with LLQ, is there are particular OS that works best and will have the patches apply cleanly.

The patch for `linux-aws-5.3-5.3.0` looks like:
```
--- ./tmp/linux-aws-5.3-5.3.0/drivers/vfio/pci/vfio_pci.c       2020-10-01 22:19:09.000000000 +0000
+++ ./tmp_old/linux-aws-5.3-5.3.0/drivers/vfio/pci/vfio_pci.c   2020-10-01 22:11:13.339678012 +0000
@@ -1239,7 +1239,12 @@
                if (ret)
                        return ret;

-               vdev->barmap[index] = pci_iomap(pdev, index, 0);
+               if (pci_resource_flags(pdev, index) & IORESOURCE_PREFETCH)
+                       vdev->barmap[index] = ioremap_wc(
+                           pci_resource_start(pdev, index),
+                           pci_resource_len(pdev, index));
+               else
+                       vdev->barmap[index] = pci_iomap(pdev, index, 0);
                if (!vdev->barmap[index]) {
                        pci_release_selected_regions(pdev, 1 << index);
                        return -ENOMEM;
@@ -1247,7 +1252,10 @@
        }

        vma->vm_private_data = vdev;
-       vma->vm_page_prot = pgprot_noncached(vma->vm_page_prot);
+       if (pci_resource_flags(pdev, index) & IORESOURCE_PREFETCH)
+               vma->vm_page_prot = pgprot_writecombine(vma->vm_page_prot);
+       else
+               vma->vm_page_prot = pgprot_noncached(vma->vm_page_prot);
        vma->vm_pgoff = (pci_resource_start(pdev, index) >> PAGE_SHIFT) + pgoff;

        return remap_pfn_range(vma, vma->vm_start, vma->vm_pgoff,
```

And for `linux-aws-5.4-5.4.0` it looks like:

```
--- a/./tmp/linux-aws-5.4-5.4.0/drivers/vfio/pci/vfio_pci.c
+++ b/./tmp_old/linux-aws-5.4-5.4.0/drivers/vfio/pci/vfio_pci.c
@@ -1500,7 +1500,13 @@ static int vfio_pci_mmap(void *device_data, struct vm_area_struct *vma)
                if (ret)
                        return ret;

-               vdev->barmap[index] = pci_iomap(pdev, index, 0);
+               if (pci_resource_flags(pdev, index) & IORESOURCE_PREFETCH)
+                       vdev->barmap[index] = ioremap_wc(
+                           pci_resource_start(pdev, index),
+                           pci_resource_len(pdev, index));
+               else
+                       vdev->barmap[index] = pci_iomap(pdev, index, 0);
+
                if (!vdev->barmap[index]) {
                        pci_release_selected_regions(pdev, 1 << index);
                        return -ENOMEM;
@@ -1508,7 +1514,10 @@ static int vfio_pci_mmap(void *device_data, struct vm_area_struct *vma)
        }

        vma->vm_private_data = vdev;
-       vma->vm_page_prot = pgprot_noncached(vma->vm_page_prot);
+       if (pci_resource_flags(pdev, index) & IORESOURCE_PREFETCH)
+               vma->vm_page_prot = pgprot_writecombine(vma->vm_page_prot);
+       else
+               vma->vm_page_prot = pgprot_noncached(vma->vm_page_prot);
        vma->vm_pgoff = (pci_resource_start(pdev, index) >> PAGE_SHIFT) + pgoff;

        /*
```

I-gor-C, 2020-10-07:
Hi @mikeb01 

Thanks for bringing this issue to our attention.
We're taking care of this and will keep you updated.

I-gor-C, 2021-01-11:
@mikeb01 The issue has been resolved, please reopen if additional assistance is required.

closed 2021-01-11 by I-gor-C