amzn/amzn-drivers #146: ena: XDP_QUERY_PROG removed from kernel 5.9, kernel patch not incorporated here

ziggythehamster, 2020-11-03:
The upstream kernel developers removed the `XDP_QUERY_PROG` command, and [patched the ena driver](https://lwn.net/ml/bpf/20200716045602.3896926-10-andriin@fb.com/) to remove support for it, but their patch isn't incorporated here. This affects all kernel 5.9s.

sameehj, 2020-11-03:
Hi @ziggythehamster,
 
We are aware of this change and already applied it to our upcoming release – 2.3.0g. It should be available within the next couple of weeks.

sameehj, 2020-11-25:
Hi @ziggythehamster ,

We have just released ENA Linux driver 2.3.0g which has a fix for the issue you mentioned above.

I'm closing this issue for now, please feel free to reopen if needed.

closed 2020-11-25 by sameehj