amzn/amzn-drivers #147: Flaky network with CPU soft lockups and ena resets

wcurry, 2020-11-09:
I opened this issue with Fedora Coreos: https://github.com/coreos/fedora-coreos-tracker/issues/665 .

We are seeing the following after attempting to bump our OS to the latest FCOS 32. I bisected the issue and found it starts in the versions below:

## Good ##

OS: Fedora Coreos 31.20200310.3.0
Kernel: 5.5.8-200.fc31.x86_64
ena version: 2.1.0K

## Bad ##

OS: Fedora Coreos 31.20200323.2.0
Kernel: 5.5.10-200.fc31.x86_64
ena version: 2.1.0K

## Good - FCOS 31.20200323.2.0 / 5.5.10-200.fc31.x86_64 ##

```
$ ssh -i ~/.ssh/k8s-dev-us-west-2 core@172.27.187.214
Fedora CoreOS 31.20200310.3.0
Tracker: https://github.com/coreos/fedora-coreos-tracker
Discuss: https://discussion.fedoraproject.org/c/server/coreos/

[core@ip-172-27-187-214 ~]$ uname -a
Linux ip-172-27-187-214 5.5.8-200.fc31.x86_64 #1 SMP Thu Mar 5 21:28:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
[core@ip-172-27-187-214 ~]$ modinfo ena
filename:       /lib/modules/5.5.8-200.fc31.x86_64/kernel/drivers/net/ethernet/amazon/ena/ena.ko.xz
version:        2.1.0K
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
srcversion:     DAAE6CFC0FC2113B5776480
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
depends:
retpoline:      Y
intree:         Y
name:           ena
vermagic:       5.5.8-200.fc31.x86_64 SMP mod_unload
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        0C:5D:ED:30:0B:3B:E3:23:0B:AD:A5:10:3E:7E:29:76:0E:6B:3A:1E
sig_hashalgo:   sha256
signature:      2A:A6:13:DB:14:78:41:12:F7:75:4D:6C:E5:B3:4E:45:6A:C0:3F:B9:
                6B:CA:73:16:A4:87:2B:42:67:D6:A5:4A:5D:1C:0F:0D:53:EF:C6:69:
                29:35:EB:AA:AC:C0:36:7F:DB:28:F7:25:1B:8E:31:A1:55:9D:78:EA:
                84:3A:61:9C:1C:58:74:AA:8B:BB:8B:AE:28:FC:9F:4D:68:CF:FA:CC:
                25:38:C6:15:F0:55:0E:7A:D1:31:CB:F9:73:C7:D4:32:C2:90:8A:10:
                31:43:BF:A1:08:12:C5:AA:96:8F:CE:F6:D0:9A:96:BA:60:18:A7:1F:
                10:1B:B2:BE:80:78:08:B0:07:14:99:E3:BD:6C:A7:D6:3E:57:45:BF:
                A4:48:6E:D4:9D:06:AE:51:C2:1C:3C:54:B0:36:8A:1D:2C:F6:0F:18:
                59:23:D9:BB:91:16:A7:EE:57:E9:7E:DB:22:0D:5D:62:25:E9:EF:97:
                F4:B7:86:DC:DE:B6:52:7C:AF:6A:CF:43:EA:A0:F7:70:D7:C5:97:8F:
                DC:7E:55:AB:F9:55:66:B8:9F:2D:C4:16:16:FE:F5:88:18:26:0E:A5:
                17:6D:64:CF:63:1A:B5:53:43:58:5D:11:19:76:4F:3D:B7:00:54:75:
                C5:45:7A:56:C7:AF:39:CF:E5:21:D0:43:58:53:20:58:09:0B:B9:AE:
                94:BE:90:51:37:DE:FF:24:74:CE:48:AB:3D:68:FB:BF:D6:5B:24:14:
                88:D9:DD:52:F3:3A:EE:6A:AA:21:77:76:C8:15:6C:50:BB:C5:21:E5:
                B5:41:C8:DA:61:61:0D:C2:48:5B:43:79:72:1D:29:94:CA:47:25:1B:
                59:AE:4D:5E:8D:5B:2C:FF:94:88:FC:34:6C:95:A8:53:8D:68:23:02:
                1D:04:A5:00:57:4F:BD:00:E4:6D:1E:1E:3A:2C:7F:43:A9:2A:3B:87:
                2A:D0:17:A8:67:74:13:A1:DA:E6:E6:8D:AA:A5:BB:4E:32:8A:67:35:
                BB:26:5C:39:9A:D9:F5:61:79:E7:E4:AE:4E:09:33:F3:F9:EE:8C:09:
                75:A6:74:1E:41:4E:82:98:A8:AA:04:99:AA:90:4D:DD:CD:CA:D8:95:
                67:2C:29:55:E8:C9:EA:23:A5:E4:EC:83:04:08:4A:CA:A0:84:1B:A1:
                4A:96:7B:3F:BD:36:2D:70:FB:A5:43:96:C3:24:69:41:A8:8E:FC:99:
                65:5F:7E:2B:A1:3D:D3:A0:77:86:F2:77:BC:69:F4:21:C7:3E:D7:89:
[... 222 lines cut ...]
[  +0.002288]  exit_mmap+0xad/0x1a0
[  +0.002330]  mmput+0x61/0x140
[  +0.002171]  do_exit+0x30d/0xb80
[  +0.002308]  do_group_exit+0x3a/0xa0
[  +0.002451]  __x64_sys_exit_group+0x14/0x20
[  +0.002804]  do_syscall_64+0x5b/0x1c0
[  +0.002538]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  +0.003437] RIP: 0033:0x55a99bc78b2b
[  +0.002415] Code: Bad RIP value.
[  +0.002271] RSP: 002b:000000c000189f58 EFLAGS: 00000246 ORIG_RAX: 00000000000000e7
[  +0.004593] RAX: ffffffffffffffda RBX: 000000c000189968 RCX: 000055a99bc78b2b
[  +0.004409] RDX: 000000c000189f90 RSI: 0000000000000240 RDI: 0000000000000000
[  +0.004430] RBP: 000000c000189f58 R08: 0000000000000009 R09: ffffffffffffffff
[  +0.004373] R10: 00007f69462c2aa0 R11: 0000000000000246 R12: 00000000000000ff
[  +0.004448] R13: 00ffffffffffffff R14: 000055a99bfd2892 R15: 0000000000000038
[  +2.030502] ena 0000:00:03.0 eth0: free uncompleted tx skb qid 0 idx 0x21e
[  +0.005300] ena 0000:00:03.0 eth0: free uncompleted tx skb qid 1 idx 0x14f
[  +0.016827] ena: ena device version: 0.10
[  +0.003123] ena: ena controller version: 0.0.1 implementation version 1
[  +0.057904] ena 0000:00:03.0: Device reset completed successfully, Driver info: Elastic Network Adapter (ENA) v2.1.0K

[  +0.228060] SELinux: mount invalid.  Same superblock, different security settings for (dev mqueue, type mqueue)
```

Thanks for your help!

wcurry, 2020-11-16:
I found the issue. We were enabling SMT on first-boot by running a service with the following command:

```
  - name: enable-smt-firstboot.service
    enabled: true
    contents: |
      [Unit]
      Description=Enable SMT on first boot on Intel CPUs to disable MDS mitigation
      DefaultDependencies=no
      Before=sysinit.target shutdown.target
      Conflicts=shutdown.target
      ConditionFirstBoot=true
      [Service]
      Type=oneshot
      ExecStart=/bin/bash -c 'active="$(cat /sys/devices/system/cpu/smt/active)" && if [[ "$active" != 1 ]] && grep -q "vendor_id.*GenuineIntel" /proc/cpuinfo; then echo "Enabling SMT." && echo on > /sys/devices/system/cpu/smt/control; fi'
      [Install]
      WantedBy=sysinit.target
```

Later reboots took advantage of another unit that appended "mitigations=auto" to kargs.

As of 31.20200323.2.0, this apparently stopped working.

When adding only "--reboot" to our kargs unit and removing the above /sys/devices... unit, our etcd cluster would not survive simultaneous immediate reboots.

I have added the following to each of the systemd service units (excluding the kargs unit) to delay their start until second-boot:

`ConditionKernelCommandLine=!ignition.firstboot`

AWSNB, 2020-11-16:
thanks for the update @wcurry 

this matches with ena output: the [  +0.004392] ena 0000:00:03.0 eth0: queue_0_tx_missed_tx: 156.  is actually indicating that the napi service that supposed to handle tx completions was not run,  possibly due to missing interrupt or napi not being scheduled for the specific core

closed 2020-11-17 by akiyano