amzn/amzn-drivers #152: Compile error with kernel 5.10.0 - CentOS 8

vvhungy, 2020-12-14:
I got a compilation error as below. Please check.
Thanks.

OS/Kernel: CentOS 8, kernel 5.10.0-1
ena driver version 2.4.0 and 2.3.0.

make: Entering directory '/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena'
make -C /lib/modules/5.10.0-1.el8.elrepo.x86_64/build M=/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena modules
make[1]: Entering directory '/usr/src/kernels/5.10.0-1.el8.elrepo.x86_64'
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_ethtool.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/dim.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/net_dim.o
/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.c: In function ‘ena_del_napi_in_range’:
/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.c:2575:3: error: implicit declaration of function ‘napi_hash_del’; did you mean ‘hash_del’? [-Werror=implicit-function-declaration]
   napi_hash_del(&adapter->ena_napi[i].napi);
   ^~~~~~~~~~~~~
   hash_del
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1805: /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.10.0-1.el8.elrepo.x86_64'
make: *** [Makefile:40: all] Error 2
make: Leaving directory '/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena'

vvhungy, 2020-12-14:
It seems that in 5.10.0, we have to use __napi_hash_del instead.

akiyano, 2020-12-14:
Hi  @vvhungy,

Thanks for the report.
Attached is a patch specifically for 5.10.
You can apply it with: git am 0001-Fix-compilation-issue-in-linux-kernel-5.10.txt
This issue will be fixed in the next driver release.

[0001-Fix-compilation-issue-in-linux-kernel-5.10.txt](https://github.com/amzn/amzn-drivers/files/5688642/0001-Fix-compilation-issue-in-linux-kernel-5.10.txt)

Thanks,
Arthur

vvhungy, 2020-12-15:
@akiyano Thanks.

akiyano, 2020-12-16:
@vvhungy,

For 5.10 you are right, netif_napi_del() translates to the code added in the patch.

However for kernel 4.4, we have not fully tested the removal of napi_hash_del() and therefore would rather you simply use the driver without this fix for kernel 4.4.

Once we release this fix in a proper version it will support both 5.10 and 4.4.

Thanks,
Arthur

pspacek, 2021-01-22:
I confirm functionality on 5.10.8.

phsung, 2021-03-03:
it seems I still have build error during MODPOST

```
ERROR: modpost: "net_dim_get_rx_moderation" [.../amzn-drivers/kernel/linux/ena/ena.ko] undefined!
ERROR: modpost: "net_dim" [.../amzn-drivers/kernel/linux/ena/ena.ko] undefined!
/.../linux_5_10/scripts/Makefile.modpost:111: recipe for target '/.../amzn-drivers/kernel/linux/ena/Module.symvers' failed
```

looks like the code is ifdef for < 5.3 but the user (in net_dim.c) still tries to use it for 5.10

akiyano, 2021-03-03:
@phsung,

If you have kernel 5.10, then all the dim code in the driver is commented out, and the dim code in the kernel should be used.
From the output you get, it seems that your 5.10 kernel does not include the dim code.
My guess is that your .config file does not include CONFIG_DIMLIB=y, which means that the dim library is not included in the kernel, which explains why these functions are not found.

Thanks,
Arthur

phsung, 2021-03-04:
Right!. Thanks!

closed 2021-04-06 by vvhungy