amzn/amzn-drivers #153: Compile error with kernel 4.4.247 - CentOS 7.9

vvhungy, 2020-12-15:
I got another compilation error as below. Please check.
Thanks.

OS/Kernel: CentOS 7.9.2009, kernel 4.4.247
ena driver version 2.4.0

make: Entering directory `/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena'
make -C /lib/modules/4.4.247-1.el7.elrepo.x86_64/build M=/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/4.4.247-1.el7.elrepo.x86_64'
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_ethtool.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/dim.o
  CC [M]  /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/net_dim.o
In file included from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.h:9:0,
                 from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_ethtool.c:8:
/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/kcompat.h:764:19: error: redefinition of ‘page_ref_count’
 static inline int page_ref_count(struct page *page)
                   ^
compilation terminated due to -Wfatal-errors.
In file included from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/dim.h:8:0,
                 from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/net_dim.c:6:
/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/kcompat.h:764:19: error: redefinition of ‘page_ref_count’
 static inline int page_ref_count(struct page *page)
                   ^
compilation terminated due to -Wfatal-errors.
In file included from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/dim.h:8:0,
                 from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/dim.c:6:
/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/kcompat.h:764:19: error: redefinition of ‘page_ref_count’
 static inline int page_ref_count(struct page *page)
                   ^
compilation terminated due to -Wfatal-errors.
In file included from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.h:9:0,
                 from /var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.c:25:
/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/kcompat.h:764:19: error: redefinition of ‘page_ref_count’
 static inline int page_ref_count(struct page *page)
                   ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_ethtool.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/dim.o] Error 1
make[2]: *** [/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/net_dim.o] Error 1
make[2]: *** [/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.4.247-1.el7.elrepo.x86_64'
make: *** [all] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/2.4.0/build/kernel/linux/ena'

akiyano, 2020-12-15:
Hi @vvhungy,

Thanks for your report.
The official kernel in CentOS 7.9.2009 is 3.10.0-1160.el7.x86_64
Is your kernel 4.4.247 taken from the upstream linux kernel repository that you built and installed yourself from sources? Or is it part of some package?

Thanks!
Arthur

vvhungy, 2020-12-15:
@akiyano,

The kernel 4.4.247 was installed from **elrepo** repository (http://elrepo.org/tiki/HomePage).

ShayAgros, 2020-12-15:
Hi @vvhungy,

Thanks for bringing it up :+1: 
I attach a patch which should fix this compilation issue.
You can apply it with: 
`git am 0001-ena-linux-Support-page_ref_count-backpropagation-fro.txt`

The patch would also be included in next driver release

[0001-ena-linux-Support-page_ref_count-backpropagation-fro.txt](https://github.com/amzn/amzn-drivers/files/5696390/0001-ena-linux-Support-page_ref_count-backpropagation-fro.txt)

Please let me know if this patch fixes your issue
Shay

vvhungy, 2020-12-16:
@ShayAgros: no error after applying the patch. Thanks.

akiyano, 2021-09-29:
Patch was released as part of version 2.5 closing

closed 2021-09-29 by akiyano