amzn/amzn-drivers #164: AF_XDP utilizes only 1/2 of available CPU cores

pspacek, 2021-01-22:
For some reason ena driver refuses to load XDP programs if number of configured queues is > 1/2 CPU cores. This leads to uneven load distribution between cores - ideally each CPU could have its own XDP queue attached to it.

This is an practical problem for high-performance DNS software which uses XDP, e.g. [Knot DNS 3.0](https://knot-dns.cz/).

Environment:
```
# uname -a
Linux ip-10-10-6-165 5.10.8-1-ec2 #1 SMP Mon, 18 Jan 2021 12:09:55 +0000 x86_64 GNU/Linux

# ethtool -i ens5
driver: ena
version: 2.4.0g
firmware-version: 
expansion-rom-version: 
bus-info: 0000:00:05.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
# ethtool -l ens5
Channel parameters for ens5:
Pre-set maximums:
RX:		n/a
TX:		n/a
Other:		n/a
Combined:	4
Current hardware settings:
RX:		n/a
TX:		n/a
Other:		n/a
Combined:	4
```

Error while loading XDP program - CLI:
```
Error: ena: Failed to set xdp program, there is no enough space for allocating XDP queues, Check the dmesg for more info.
```

Error while loading XDP program - dmesg:
```
ena 0000:00:05.0 ens5: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (4), the maximal queue count (4)
```

An obvious workaround is to use
```
ethtool -L $ETH combined $(( $(nproc) / 2))
```
but it under-utilizes instance cores.

Can you please remove this limitation?

Thank you!

sameehj, 2021-01-25:
Hi @pspacek,

Please contact me at sameehj@amazon.com to further continue the investigation.

pspacek, 2021-01-26:
Sent yesterday. Did you receive my message, or did it end up in spam?

sameehj, 2021-02-03:
Investigation completed offline. Closing.

closed 2021-02-03 by sameehj

pspacek, 2021-02-16:
Info for people who find this via search engine: The issue was confirmed in current versions and is known. Contact support if you have performance trouble because of this.