amzn/amzn-drivers #166: "rx_descriptor_status" function not implemented in DPDK ENA pmd

subhrch, 2021-02-07:
Hi,

I found that in case of DPDK ENA pmd "rx_descriptor_status" function pointer is not implemented unlike other PMDs in dpdk. We have a dpdk application which heavily rely on this function before calling any rx_burst function. We don't wait on a busy loop to continuously check on the rx_burst function and that is a crucial part of the application design. The rx packet processing only happens when "rx_descriptor_status" returns DD.  We wanted to run our application in AWS EC2 with ena driver for better performance but lack of this is prohibiting from doing so. Can you please help and let me know if this functionality will be available in any near future. Is it feasible to implement this functionality and back port to older DPDK releases.

If this is not the right forum to get an answer please help me with a link where I can get some help w.r.t this problem in ENA DPDK pmd.

thanks
Subhransu

I-gor-C, 2021-02-07:
Hi Subhransu,

Thanks for reaching out. I'll check the feasibility of your request and get back to you.

I-gor-C, 2021-02-08:
Appears to be feasible. I'll update on the progress.

subhrch, 2021-02-12:
That is great. Thanks @I-gor-C . I would be waiting to try it.

subhrch, 2021-02-18:
Hi @I-gor-C . How are you ? 
Just wanted to check if there will be any tentative time when I can expect the changes for try out.  Also what will be the plan for the final change being available in DPDK LTS releases ? We are currently planning to migrate to DPDK 20.11 LTS.

I-gor-C, 2021-02-18:
Hi @subhrch 
I currently don't have a timeline but we will try to make it available for the next DPDK release - I'll update if there are changes on that. I don't see a problem to backport it to 20.11 once the feature is ready.

I-gor-C, 2021-04-30:
Hi @subhrch 
This feature won't make it to the v21.05 release.
We might be able to provide you with a prerelease version if it's required urgently.
Please contact me directly at igorch@amazon.com if you're interested.

rodgarrison, 2022-03-15:
OP is referring to:

- [rte_eth_rx_descriptor_status](https://doc.dpdk.org/api/rte__ethdev_8h.html#ad3129ac553f76fd90d7e5eea46a63398)

In this comment I am referring to the TX equivalent immediately below in link.

Please note as of this writing `rte_eth_tx_descriptor_status()` returns -ENOTSUP on DPDK head commit (917229c24e871bbc3225a0227eb3f0faaa7aaa69 @ Fri Mar 4 16:24:24 2022 +0000).

Unless the driver team can recommend an alternative it's imperative both the RX and TX APIs are implemented. For the TX side consider the following inevitable use-case:

1. A TXQ lcore has rte_eth_tx_burst all the packets it wants to send
2. The TXQ lcore, being done, exits its thread returning 0
3. Back on the main lcore the process blocks until `rte_eal_mp_wait_lcore` is done
4. The process collects stats including especially the number of packets sent. **Naively** we'd expect the number of out packets reported by either  rte_eth_stats_get or rte_eth_xstats_get would equal the number of successfully enqueued packets reported by rte_eth_tx_burst. **In fact this will not happen in general**.

Without the ability to confirm all the descriptors in the TXQs rings are DONE via rte_eth_tx_descriptor_status before (2) returns, the process cannot exit clean knowing everything it intended to send was in fact sent. Put another way, there will in general be work in-queue on the HW still in-progress.

shaibran, 2022-06-06:
update, the feature is postponed to next PMD 2.8.0 release