amzn/amzn-drivers #168: RX interrupt support in the DPDK ENA PMD

mhawari, 2021-03-03:
Hi,

I noted that the functions `rx_queue_intr_enable` and `rx_queue_intr_disable` were not implemented in `ena_dev_ops`. This means that the ENA PMD can't be used in interrupt mode, contrary to other PMDs. Are there any plan to implement RX interrupt support in the near future?

Thanks and best regards
Mohammed

I-gor-C, 2021-03-03:
Hi Mohammed,

Let me check this and get back to you

I-gor-C, 2021-03-03:
@mhawari Interrupts are usually not used with PMDs because of DPDK's general concept to use polling mode. Could you please provide some details about your usecase?

mhawari, 2021-03-03:
Thanks for the quick answer @I-gor-C ! Here are some details. Interrupt mode for ENA is extremely useful to the Calico-VPP project [1]. It relies on VPP [2] to provide a user-space, fast data plane to Calico. When deployed in AWS, VPP have access to the underlying ENA either by using DPDK or AF_XDP, in polling or interrupt mode. DPDK provides better performance than AF_XDP for a lot of use cases, because it supports TSO, while AF_XDP does not. Interrupt mode support with DPDK would considerably improve resource-efficiency in a lot of scenarios, as we wouldn't need to dedicate CPU cores to busy polling anymore. This topic is covered with more details here [3] 

[1] https://github.com/projectcalico/vpp-dataplane
[2] https://fd.io/vppproject/vpptech/
[3] https://fosdem.org/2021/schedule/event/sdn_calicovpp/

mhawari, 2021-03-03:
CCing @mdr78

jtollet, 2021-03-17:
Hello, Is there any update on this one ?

I-gor-C, 2021-03-18:
Hi @jtollet 
We're still checking the feasibility of this. I'll update in a few days.

I-gor-C, 2021-03-18:
We're planning to deliver this feature to the DPDK 21.05 release.
I'll update if the timeline shifts.

jtollet, 2021-04-08:
Hello @I-gor-C, I'd be interested to test it ahead of time if you have a pre-release. Feel free to ask if that makes sense to you.

jtollet, 2021-04-23:
> We're planning to deliver this feature to the DPDK 21.05 release.
> I'll update if the timeline shifts.

Hello @I-gor-C, can you confirm it will be part of DPDK 21.05 ?

cdtomkins, 2021-05-06:
@I-gor-C Can we get an update on whether this will be in DPDK 21.05, please?

I-gor-C, 2021-05-06:
Hi @cdtomkins 
Unfortunately, it won't make it to 21.05 and is pushed to 21.08. We can share the prerelease if you would like to try it in the meanwhile.

cdtomkins, 2021-05-06:
Thanks @I-gor-C, @jtollet will probably want that.

I-gor-C, 2021-05-06:
Please contact me directly at igorch@amazon.com

jtollet, 2021-05-06:
Definitely helpful. We’ll give it a try and will let you know !

Répondre à : amzn/amzn-drivers ***@***.***>
Date : jeudi 6 mai 2021 à 15:07
À : amzn/amzn-drivers ***@***.***>
Cc : Jerome Tollet ***@***.***>, Mention ***@***.***>

jtollet, 2021-05-21:
Hello, I can confirm that according to our first test, Interrupt mode now works fine. Thanks a lot!

AloysAugustin, 2021-06-09:
Hi @I-gor-C , thanks for sharing the patch privately with us. We've been using it for a bit now without issues in EKS.

We'd like to include it in the next Calico VPP dataplane release, but to do so, we'd need to make it public in the VPP gerrit (so that we can patch the DPDK version we use, while the patch is not upstreamed in DPDK). Would you have any objection to that?

ndagan, 2021-06-10:
Hi @AloysAugustin,
Yes, you are good to go to post it in VPP gerrit.  We plan to release it for DPDK v21.08.

AloysAugustin, 2021-07-13:
Hello, it looks like some changes in dpdk 21.05 cause the patch that was provided not to apply anymore, and I didn't see an equivalent patch merged in the dpdk main branch. Would it be possible for you to share an updated version of the patch that would apply to dpdk 21.05 or to the main branch ?
Thanks,
Aloys

semihalf-krawczyk-michal, 2021-07-14:
Hi @AloysAugustin,

patch with Rx interrupts was just pushed for the community review and if there won't be objections, it will be merged for the DPDK v21.08: https://patches.dpdk.org/project/dpdk/patch/20210713154118.32111-5-mk@semihalf.com/.

The patch was modified a bit since the last version you got (especially the error handling path for the admin interrupt), but the core functionality should remain the same.

You can try to cherry-pick it from there. It should be applicable to the DPDK v21.05 as well.

onong, 2021-08-03:
Hi @semihalf-krawczyk-michal,

Just wanted to let you know that we have tested the patch with dpdk 21.05 in calico/vpp and it seems to be working fine. Look forward to the patch being merged in 21.08.

Thanks,
Onong

semihalf-krawczyk-michal, 2021-08-04:
Hi @onong,

Thanks a lot for your feedback!

The another good news is that this feature was already merged last week and can be found in tag v21.08-rc2 and newer, so it will be part of the DPDK 21.08 release.

Thanks,
Michal

onong, 2021-08-04:
Thats great news, thanks much! @semihalf-krawczyk-michal

shaibran, 2021-12-14:
@onong, fix was merged to  DPDK 21.08

closed 2021-12-23 by akiyano