amzn/amzn-drivers #169: RSS hash result doesn't match Toeplitz

soulofmitsui, 2021-04-02:
Hi,
I'm using DPDK on ENA 2.4.0g. The rss hash returned by ENA doesn't math Toeplitz. Can you please confirm the hash key ENA uses is {0x6d5a56da, 0x255b0ec2, 0x4167253d, 0x43a38fb0, 0xd0ca2bcb, 0xae7b30b4, 0x77cb2da3, 0x8030f20c, 0x6a42b73b, 0xbeac01fa}? Does the hash input include src_addr, dst_addr, dst_port, and src_port?

I-gor-C, 2021-04-02:
@soulofmitsui ENA driver in DPDK currently supports only CRC32 hashing. Reconfiguration isn't enabled. For TCP/UDP the hash input indeed includes L3 src/dst addresses and L4 src/dst ports.

soulofmitsui, 2021-04-02:
@I-gor-C, Thanks for reply. Are you sure it's CRC32? Multiple tickets (include #109) suggest ena uses Toeplitz. The PMD driver appears to setup CRC32 but it's failed with EOPNOTSUP.
https://github.com/amzn/amzn-drivers/issues/136
https://github.com/amzn/amzn-drivers/issues/135

I-gor-C, 2021-04-03:
Let me check this closer. Will update here.

soulofmitsui, 2021-04-05:
Appreciate for the help @I-gor-C. Even if ENA DPDK driver doesn't currently support RSS configuration, I'd like to know the built-in hash key assuming ENA uses Toeplitz. By replicating the hash function and key, my code can control which Rx queue on the target will receive packets with specific ports.

I-gor-C, 2021-04-05:
@soulofmitsui Yes, the ENA DPDK driver doesn't currently support RSS reconfiguration. Also, as you have observed as well, the initial RSS hash configuration to CRC32 fails on initialization. ENA will use Toeplitz hashing with the key as described in https://github.com/amzn/amzn-drivers/issues/109

soulofmitsui, 2021-04-05:
@I-gor-C, I tried with the Toeplitz implementation in DPDK with the key in #109. I found the hash result from ENA doesn't match DPDK. That's why I suspected ENA uses a different key or different hash inputs in the first place. For example, for src 10.40.99.67:12011 and dst 10.40.113.88:5001, ENA hash result is 0x499f224d and DPDK Toeplitz gives 0xdb7dcc53. I trust the DPDK result because it matches the verification suite https://www.verical.com/datasheet/intel-ethernet-controller-jl82599eb-s-r1w7-1027643.pdf which uses the same key.

Can you please double check the ENA Toeplitz implementation is standard? Also what's the byte sequence of L3 addresses and L4 ports? DPDK uses the following:

struct rte_ipv4_tuple {
    uint32_t    src_addr;
    uint32_t    dst_addr;
    union {                   
        struct {
            uint16_t dport;
            uint16_t sport;
        };
    }; 
};

nafeabshara, 2021-04-05:
@soulofmitsui

Could this be related to the symmetry of the hash function.   Before applying the hash, one need to order the src/dst address from smaller to larger, and re-order dport/sport from smaller to larger.    This attribute enables same “symmetric” hash function so Tx and Rx direction of same flow have same hash

soulofmitsui, 2021-04-06:
@nafeabshara, I've tried different ordering of dst/src addr and dst/src ports. None of them match the ENA hash result.

Does ENA reorder the bytes like you suggested? I thought Topelitz uses special key (https://www.ran-lifshitz.com/2014/08/28/symmetric-rss-receive-side-scaling/) to achieve symmetricity.

zorikm, 2021-04-06:
@soulofmitsui ,
The standard Toeplitz calculation can be found in the following  Python implementation online: https://gist.github.com/joongh/16867705b03b49e393cbf91da3cb42a7

For the network accelerated instance types (as the one you are using) Toeplitz calculation differs from the one above in the following:
 
1. The hash key is passed to it in reverse byte order.
2. Calculation begins with an initial result value of 0xffffffff.

This results in the following modifications :

@@ -9,11 +9,11 @@ KEY=[]
 def reset_key():
     global KEY
     KEY = [
-        0x6d, 0x5a, 0x56, 0xda, 0x25, 0x5b, 0x0e, 0xc2,
-        0x41, 0x67, 0x25, 0x3d, 0x43, 0xa3, 0x8f, 0xb0,
-        0xd0, 0xca, 0x2b, 0xcb, 0xae, 0x7b, 0x30, 0xb4,
-        0x77, 0xcb, 0x2d, 0xa3, 0x80, 0x30, 0xf2, 0x0c,
-        0x6a, 0x42, 0xb7, 0x3b, 0xbe, 0xac, 0x01, 0xfa,
+        0xfa, 0x01, 0xac, 0xbe, 0x3b, 0xb7, 0x42, 0x6a,
+        0x0c, 0xf2, 0x30, 0x80, 0xa3, 0x2d, 0xcb, 0x77,
+        0xb4, 0x30, 0x7b, 0xae, 0xcb, 0x2b, 0xca, 0xd0,
+        0xb0, 0x8f, 0xa3, 0x43, 0x3d, 0x25, 0x67, 0x41,
+        0xc2, 0x0e, 0x5b, 0x25, 0xda, 0x56, 0x5a, 0x6d,
         0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
         0x00, 0x00, 0x00, 0x00
     ]

and
@@ -38,7 +38,7 @@ def shift_key():
 def compute_hash(input_bytes, N):
     reset_key()
-    result = 0;
+    result = 0xffffffff;
Please let us know if this resolves your issue.
We will add this information to Readme

soulofmitsui, 2021-04-06:
@zorikm, Unfortunately it still doesn't match. Here is the change I made to toeplitz_based_hash.py:
```
12,16c12,16
<         0x6d, 0x5a, 0x56, 0xda, 0x25, 0x5b, 0x0e, 0xc2,
<         0x41, 0x67, 0x25, 0x3d, 0x43, 0xa3, 0x8f, 0xb0,
<         0xd0, 0xca, 0x2b, 0xcb, 0xae, 0x7b, 0x30, 0xb4,
<         0x77, 0xcb, 0x2d, 0xa3, 0x80, 0x30, 0xf2, 0x0c,
<         0x6a, 0x42, 0xb7, 0x3b, 0xbe, 0xac, 0x01, 0xfa,
---
>         0xfa, 0x01, 0xac, 0xbe, 0x3b, 0xb7, 0x42, 0x6a,
>         0x0c, 0xf2, 0x30, 0x80, 0xa3, 0x2d, 0xcb, 0x77,
>         0xb4, 0x30, 0x7b, 0xae, 0xcb, 0x2b, 0xca, 0xd0,
>         0xb0, 0x8f, 0xa3, 0x43, 0x3d, 0x25, 0x67, 0x41,
>         0xc2, 0x0e, 0x5b, 0x25, 0xda, 0x56, 0x5a, 0x6d,
41c41
<     result = 0;
---
>     result = 0xffffffff;
74a75
> example3 = ('10.40.99.67', '10.40.113.88', 12011, 5001)
85a87,93
> if compute_hash(get_input(*example3), 12) == 0x499f224d:
>     print 'example3: Hash OK'
> else:
>     print 'example3: Hash NOT OK'
> 
> result = compute_hash(get_input(*example3), 12)
> print 'example3:' + hex(result)

```
The result is 0xcf103cb5 which doesn't match 0x499f224d from ENA DPDK:
```
example1: Hash NOT OK
example2: Hash NOT OK
example3: Hash NOT OK
example3:0xcf103cb5

```
I'm using UDP instead of TCP, not sure if that changes anything. Here is the tcpdump output:
`02:01:30.405413 IP 10.40.99.67.12011 > 10.40.113.88.commplex-link: UDP, length 8192
`
I got the DPDK ENA hash result by printing hash.rss of rte_mbuf:
```
        struct rte_mbuf *rtem = bufs[i];
        printf("rss hash %x\n", rtem->hash.rss);
```
`rss hash 499f224d
`

zorikm, 2021-04-07:
Thanks @soulofmitsui 
We are looking into it.

I-gor-C, 2021-04-08:
Hi @soulofmitsui 

We identified that setting the key in DPDK driver is different form the Linux one, and we are working on the fix that would make it the same.
In the mean time - here is how you need to modify the key to get the result you are looking for:

Please take the standard Toeplitz implementation. E.g. as in https://gist.github.com/joongh/16867705b03b49e393cbf91da3cb42a7
and replace the key with this one:
```
        0xbe, 0xac, 0x01, 0xfa, 0x6a, 0x42, 0xb7, 0x3b,
        0x80, 0x30, 0xf2, 0x0c, 0x77, 0xcb, 0x2d, 0xa3,
        0xae, 0x7b, 0x30, 0xb4, 0xd0, 0xca, 0x2b, 0xcb,
        0x43, 0xa3, 0x8f, 0xb0, 0x41, 0x67, 0x25, 0x3d,
        0x25, 0x5b, 0x0e, 0xc2, 0x6d, 0x5a, 0x56, 0xda,
```
This is the original key with bytes reversed and then the bytes in each 4-byte-word reversed again.
No need to change the initial result value from 0 to 0xffffffff.
With the above steps we were able to recreate the result that you are getting from your DPDK application.

soulofmitsui, 2021-05-14:
Hi,

A follow up question. DEV_RX_OFFLOAD_RSS_HASH isn't set in rx_offload_capa returned by rte_eth_dev_info_get() from DPDK ENA driver. Is that a mistake? Clearly ENA supports rss offload.

I-gor-C, 2021-05-15:
Yes, it's a mistake and it's fixed in the upcoming DPDK v21.05 release. The patch is available @ http://git.dpdk.org/dpdk/commit/?id=b418f0d29977f031e18185d78198f2717bcfb361

soulofmitsui, 2021-05-15:
Thanks @I-gor-C.

soulofmitsui, 2021-06-02:
Hi,

A follow up question regarding the Toeplitz key. As you explained you will make a change to unify the Toeplitz key between DPDK and Linux, will the DPDK key change as a result? If yes, that would break my code which assumes the key currently used by DPDK driver. Will I be able to query the key at runtime when the change happens?

shaibran, 2021-06-02:
> Hi,
>
> A follow up question regarding the Toeplitz key. As you explained you will make a change to unify the Toeplitz key between DPDK and Linux, will the DPDK key change as a result? If yes, that [...]

Hey @soulofmitsui, we indeed plan to change the RSS key, but once we introduce this change, we will also provide a way to query it at runtime

soulofmitsui, 2021-06-02:
Thanks @shaibran. I couldn't find where the current RSS key is set in the DPDK driver. Is changing the RSS key independent of the DPDK driver? In that case, if my code still links to the current DPDK driver when the RSS key changes, I will not be able to query the new key. I'm trying to find if my code can work future proof. Thanks!

shaibran, 2021-06-06:
@soulofmitsui currently the driver binds to a default key which is used by the device. Once we introduce the ability to use a different key, we will also provide API for the application to probe the device for the used key

yongninghu, 2021-11-12:
@shaibran 
Hi, I am using the dpdk 21.08 on my app to test out the RSS hash key reconfiguration,
I used
rte_eth_dev_rss_hash_update
to set the hash key, the return code is 0, but I see an unsupported message:
ena_rss_hash_set(): Setting RSS hash fields is not supported. Using default values: 0xc30

subsequently, I tried to retrieve the hash key using:
rte_eth_dev_rss_hash_conf_get

It returned: -95.
Is this expected behavior? Thanks.

zorikm, 2021-11-12:
@yongninghu 
What instance type are you using?

yongninghu, 2021-11-12:
@zorikm m5dn.4xlarge

yongninghu, 2021-11-15:
@zorikm Hi, bumping post. 
For debugging,
Erred out on ena_set_hash_fields.
rss_hf = ETH_RSS_NONFRAG_IPV4_UDP
`
	for (i = 0; i < ENA_ADMIN_RSS_PROTO_NUM; i++) {
		rc = ena_com_fill_hash_ctrl(ena_dev,
			(enum ena_admin_flow_hash_proto)i,
			selected_fields[i].fields);
		if (unlikely(rc != 0)) {
			PMD_DRV_LOG(DEBUG,
				"Failed to set ENA HF %d with fields %" PRIu16 "\n",
				i, selected_fields[i].fields);
			return rc;
		}
	}
`
Failed on:
i = 0.
selected_fields[i].fields = 0.

shaibran, 2021-11-16:
@yongninghu thank you for the input, I will investigate this

shaibran, 2021-11-22:
@yongninghu  Hey, this is expected behavior, since it depends on the HW support.
You can find a detailed response for similar question on the mailing list: http://mails.dpdk.org/archives/users/2021-November/005989.html (DPDK uses common structure for setting hash fields and the hash key, so it can be configured using one function call. PMD cannot support setting RSS hash key without supporting setting RSS hash fields)

shaibran, 2021-11-28:
@yongninghu We were not able to reproduce same error in house using key modification on the same instance type, thus we suspect this to be application related. You are welcome to contact me through mail and I'll do my best to assist

yongninghu, 2021-11-29:
@shaibran can you post the your test application code on github? I am wondering if I am missing any pre-required configuration steps.  What is the email that I can contact you at? Thanks.

shaibran, 2021-11-30:
@yongninghu sure, feel free to email me at shaibran@amazon.com, we can discuss it there; I will provide the patch and the instructions via mail

closed 2021-12-23 by akiyano