amzn/amzn-drivers #170: XDP_REDIRECT packets being dropped somewhere

vulcanfan, 2021-05-05:
Hi,

I have Amazon Linux 2 with the 5.4 kernel uplift, and ENA driver 2.5.0g built and installed on a d3.2xlarge instance.  3 network instances, eth0, eth1 and eth2, with XDP programs on eth1 and eth2.  The intent is to forward, using XDP_REDIRECT, packets that ingress on eth1 out on eth2 (with some manipulation etc).
I have reduced the effective queues using 'ethtool -L' on each interface to 'less than half' of the available queues, including trying 'combined 1'.

An ingress packet on eth1 clocks up the 'queue_0_rx_xdp_redirect' statistic, and on eth2 I see the 'queue_1_xdp_tx_cnt' statistic increment too.  But the Tx packet count on eth2, as seen with 'ifconfig', does not increment, and I don't see the packet where it was supposed to go.

Is there something I have missed?  What I don't understand, if someone could help in any way, is why the outbound redirected packet appears on what seems to be 'queue 1' for eth2 even though I had reduced the combined queues to 1, so only queue 0 presumably is what the NIC is actually reading from.  What does the 'queue_xx_xdp_tx_cnt' actually mean when the queue numbers that statistic relates to are the ones 'freed' by the obligatory 'ethtool -L' queue reduction for XDP.  Or more to the point, how could I ensure the XDP_REDIRECT packet is scheduled on queue 0 here, if that's what eth2 is reading from now?

Many thanks in advance

ShayAgros, 2021-05-06:
Hi, thanks for reaching out.

#### Regarding the issue itself
I don't seem to be able to reproduce the issue. I only  tested kernel 5.8, since on kernel 5.4 there is some discrepancy between the Clang version I have an the kernel version.

What program do you use to redirect the traffic, what Clang version and what is the output of `sudo cat /sys/kernel/debug/tracing/trace` while running the XDP program?
Also do you see any errors from the driver when running `sudo dmesg` ?

#### Regarding your general questions, this is some overview:

##### queue naming

The driver's queue naming is indeed confusing. When the ENA driver loads an XDP program, it dedicates
* the first half of the channels (queue pairs) to RX-TX traffic, where the TX traffic can come solely from the kernel (e.g. sockets)
* The second half of the channels only have a TX queue. These queues are used to transmit packets that are redirected using XDP (either XDP TX or XDP REDIRECT action).

We thought that adding the word `xdp` to the queue description in `ethtool -S` would highlight that the TX queue serves only XDP traffic. This can be deducted from the channel number as well.
For example on 8 channels setup, after loading an XDP program, channels 0-3 would have RX-TX queues and channels 4-7 would only have TX queue to process packets redirected using XDP. So when running `ethtool -S [interface name]` on this instance the stats would look something like this:
```
```
##### How to debug the root cause of the issue

The RX queues with XDP program loaded have stats which indicate the result of the XDP program. The counters are `xdp_drop, xdp_pass, xdp_tx, xdp_invalid, xdp_redirect`. To bisect the issue one can do use the following information:
1. Does the RX queue receives any traffic at all in the first place ? To verify if check the `cnt` counter of each RX queue. It should increase for every packet regardless of XDP program return value
2. Does the XDP program return the correct action ? To verify it check the `xdp_*` counters on the RX queue. If the packet is being redirected with XDP_REDIRECT you should see `xdp_redirect` counter increases. (by the symptoms you describe it might be that the program returns XDP_TX since the xmit counter on the same interface increases)
3. It the packet being transmitted on the destination interface ? To verify it check the counters with the `xdp_tx` word in them in the destination interface (`eth2` in your test). The counters that should increase are either `cnt` which indicates that the packet was sent successfully (not to be confused with it being valid. The device would drop it it the headers don't make sense) or `prepare_ctx_err` which indicates a problem (usually the TX queue doesn't keep up with the redirection rate).

Other than these counters check if some counters' change doesn't make sense. For example, `xdp_invalid` or any counter with the word `err` are a red flag.

For your convenience I attach a small wrapper script I wrote for `ethtool -S` which polls the stats every second and only displays the stats that have changed from previous iteration. It also sums up the queue specific values and shows their total:
usage:
`./diff_stats.sh [interface name]`
(to show only the sum on all channels without displaying the per-queue statistic. Useful to get an overview of the interface)
`./diff_stats.sh [interface name] | grep -v queue_`

(remove the .txt extension)
[diff_stats.sh.txt](https://github.com/amzn/amzn-drivers/files/6433710/diff_stats.sh.txt)

vulcanfan, 2021-05-06:
Hi,

Thanks very much for your comprehensive reply and post.

In answer to your initial questions:  
 a)  There is nothing in dmesg;  
 b) The XDP/BPF program on eth1 is our own.   I can verify what it does using the BPF test function and also forcing it to do XDP_PASS, in which case the ingress packets (after modification) end up in the kernel after all and I can view them with tcpdump -i eth1.  Although not sure what the NIC actually does to validate an outbound packet?  The src MAC address is set to eth2 in my case, the source NIC.  Dest MAC and IP set to a different EC2 instance.  The source IP varies, as this is a routing function.  The IP checksum is correct (well, checked off-server), although the NIC may do it itself.
 c)   There is nothing in /sys/kernel/debug/trace/, apart from our own debug printk statements.

There is an XDP program on eth2 for ingress packets on eth2, but that shouldn't make a difference to packets egressing on eth2?

Running the script that you kindly provided produces the following when applied to eth2.  With an identical packet ingressing on eth1 every second the script output is the same every interval:
```
===============
time diff is 1.006789838
queue_1_xdp_tx_cnt:  1
queue_1_xdp_tx_bytes:  558
queue_1_xdp_tx_napi_comp:  1
queue_1_xdp_tx_tx_poll:  1
queue_1_xdp_tx_doorbells:  1
queue_1_xdp_tx_unmask_interrupt:  1
ena_admin_q_submitted_cmd:  1
ena_admin_q_completed_cmd:  1
total_xdp_tx_cnt:  1
total_xdp_tx_bytes:  558
total_xdp_tx_napi_comp:  1
total_xdp_tx_tx_poll:  1
total_xdp_tx_doorbells:  1
total_xdp_tx_unmask_interrupt:  1
```

So that seems to claim that there is a packet txed on eth2 every second, but still the Tx packet count doesn't increment:
```
eth2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 3800
        inet 10.0.0.233  netmask 255.0.0.0  broadcast 10.255.255.255
        inet6 fe80::831:29ff:fee6:eaf1  prefixlen 64  scopeid 0x20<link>
        ether 0a:31:29:e6:ea:f1  txqueuelen 1000  (Ethernet)
        RX packets 301  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 261  bytes 23730 (23.1 KiB)                        <--- not incrementing despite queue_1_xdp_tx_cnt incrementing
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
Should I expect 'TX packets' to increment as the NIC actually places a packet on the wire when queue_1_xdp_tx_cnt increments?

Many thanks for your assistance!

vulcanfan, 2021-05-14:
Hi,

Many thanks for the patch.  I confirm that the output of 'ifconfig eth2' now increments with the 'ethtool -S eth2' xdp_tx values.

So I still can't see sight of the outbound packet on-the-wire though for some reason.  Neither by checking where the packet is meant to go (open security), or by trying the AWS packer Mirror function to see it.

I'm happy to replicate your own environment, I'm not wedded to Ubuntu 20.04 at all.  Would that be possible, could you describe the OS etc that you are using please?

Many thanks in advance!

ShayAgros, 2021-05-19:
Hi @vulcanfan sorry for the late response.

I created a "dirty" XDP_REDIRECT program that works. My setup looks something like this:
```
instance1       instance2                instance3
┌───────┐       ┌────────────────┐       ┌────────┐
│       │       │                │       │        │
│ ┌─────├──────►├─────┬───►┌─────├──────►├─────┐  │
│ │     │       │     │    │     │       │     │  │
│ │eth1 │       │eth2 │    │ eth3│       │eth4 │  │
│ └─────┤◄──────├─────┘◄───┴─────┤◄──────┼─────┘  │
│       │       │                │       │        │
└───────┘       └────────────────┘       └────────┘
```

Where:
- `eth1` and `eth2` are in the same subnet
- `eth3` and `eth4` are in the same subnet, *which is different* than the subnet of `eth1`/`eth2`
- instance 2 runs `Ubuntu Server 20.04 LTS` which is available through quick-launch. For example I used `ami-03d5c68bab01f3496` in Oregon (us-west-2), but I don't think it matters as long as you're using Ubuntu 20.04.
- instance 1 and instance 3 can run any OS, and can be the same instance (i.e. a single instance with two interfaces that belong to different subnets)

The test I used is to ping `instance 2` from `eth1` and verifying that it arrives to `eth3`. The ICMP request is being replied by `instance 3`. The reply is receivedback on `eth1`.
(Similar to the diagram)

These are the steps I took:
- brought up the three instances
- connected to `instance 2`
  - Install dependencies:
    ```
       sudo apt-get update
       sudo apt-get install make gcc clang-11 libelf-dev pkg-config
     ```
   - export envars
     ```
     export CLANG=clang-11
     export LLC=llc-11
     ```
   - Download XDP REDIRECT sample program
     [xdp_tests.tar.gz](https://github.com/amzn/amzn-drivers/files/6508320/xdp_tests.tar.gz) and run
       ```
       tar xvzf ./xdp_tests.tar.gz
       cd ena-xdp
       make deps
       make
       ```
   - Run the XDP Redirect program with
     ```
     cd redirect
     sudo ./xdp_redirect_user -f eth2 -s eth3 -r [ipv4 address of eth4]
     ```
     e.g. on my setup the command is `sudo ./xdp_redirect_user -f ens6 -s ens7 -r 172.31.68.114` 
- connect to `instance 1` and ping `eth2`. You should see the ICMP requests get replied and the XDP program printing that the packets are being redirected.

Please let me know if it works for you as well, or if you need any assistance with these steps.

mwiget, 2021-07-15:
I ran into the very same issue, while moving an XDP application onto aws ec2 instances and https://github.com/amzn/amzn-drivers/issues/170#issuecomment-844026592 helped me tremendously in isolation the root cause of my problem. I got the sample program working on **intel** based ubuntu 20.04 ec2 instances (c5n.large, c5n.xlarge) in us-west-2 and us-east-2 after installing the lastest ena driver from this repo.

The sample program doesn’t work on ARM based instances like c6gn.large and c6gn.xlarge in us-east-2 (haven’t tried this in any other region, don’t think it matters).

Steps taken to install the ena driver (doing this as part of a cloud-init script):

```
wget https://github.com/amzn/amzn-drivers/archive/master.zip
unzip master.zip     
cd amzn-drivers-master/kernel/linux/ena/ && make && cp ena.ko /lib/modules/$(uname -r)/ && depmod && rmmod ena && modprobe ena
```

and to install and run the provided sample app:

```
wget https://github.com/amzn/amzn-drivers/files/6508320/xdp_tests.tar.gz
tar xvf xdp_tests.tar.gz
export CLANG=clang-11
export LLC=llc-11
cd ena-xdp
sed -i s/x86_64/aarch64/ common/common.mk      # required to compile on ARM based instances
make deps
make

cd redirect
sudo ./xdp_redirect_user -f ens7 -s ens6 -r 10.0.1.101
```

Driver and kernel info:

```
mwiget@awsr1:~/ena-xdp/redirect $ sudo ethtool -i ens7
driver: ena
version: 2.5.0g
firmware-version:
expansion-rom-version:
bus-info: 0000:00:07.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes

mwiget@awsr1:~/ena-xdp/redirect$ uname -a
Linux awsr1 5.8.0-1038-aws #40~20.04.1-Ubuntu SMP Thu Jun 17 13:20:15 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

The sample app only reports redirecting the outgoing icmp packet and that packet never makes it to the remote instance:

```

iget@awsr1:~/ena-xdp/redirect$ sudo ./xdp_redirect_user -f ens7 -s ens6 -r 10.0.1.101
ens7: changing channels nr from 4 to 2
ens6: changing channels nr from 4 to 2
Packets would be redirected from if ens7 (ix 4) to if ens6 (ix 3), and
the other way around

Second ip addr: 0x6401000a, redirection ip addr: 0x6501000a
libbpf: elf: skipping unrecognized data section(8) .rodata.str1.1
Forwarded 940597837843 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
Forwarded          1 pkt/s on both interfaces
```

On Intel based instances, it reports `2 pkt/s on both interfaces` and ping works.

@ShayAgros: Shall I create a separate issue to report ARM related XDP_REDIRECT problem?

mwiget, 2021-07-15:
Looks like the ena.ko module faces an exception every time a packet gets redirected. This only happens when running this on an ARM based instance, in my case c6gn.large.

`ens6: Failed to create and store tx meta desc`

ens6 is the egress interface for the icmp request coming in on ens7.

```
Jul 15 08:22:27 awsr1 /usr/sbin/irqbalance: IRQ ens7-Tx-Rx-0(53) guessed as class 0
Jul 15 08:22:27 awsr1 /usr/sbin/irqbalance: IRQ ens7-Tx-Rx-1(54) guessed as class 0
Jul 15 08:23:40 awsr1 systemd-networkd[458]: ens5: DHCPv6 address 2600:1f16:2a8:2201:289d:37ba:7281:cca5/128 timeout preferred 150 valid 450
Jul 15 08:24:45 awsr1 kernel: [ 1518.215878] ena 0000:00:06.0 ens6: Failed to create and store tx meta desc
Jul 15 08:24:46 awsr1 kernel: [ 1519.006020] ena 0000:00:06.0 ens6: Trigger reset is on
Jul 15 08:24:46 awsr1 kernel: [ 1519.006595] ena 0000:00:06.0 ens6: tx_timeout: 0
Jul 15 08:24:46 awsr1 kernel: [ 1519.007088] ena 0000:00:06.0 ens6: suspend: 0
Jul 15 08:24:46 awsr1 kernel: [ 1519.007552] ena 0000:00:06.0 ens6: resume: 0
```

ShayAgros, 2021-07-18:
@mwiget  sorry for the late response.
First let me thank you for debugging the issue (:

For some instance sizes (e.g. c6gn) we need to create a separate meta descriptor (containing some information about the packet) for every packet we send. The driver "forgot" to set the right flag to reflect that

Probably a more thorough fix is needed, but your fix actually does the trick. Alternatively you can use
[annotate_valid_meta.patch.zip](https://github.com/amzn/amzn-drivers/files/6836460/annotate_valid_meta.patch.zip)
(apply it with `git am annotate_valid_meta.patch.zip`)

which aligns the driver's handling of XDP xmitted packets with non-XDP code path (the effect is the same as your patch though, so your change can be used instead).

Next driver version will contain the fix. Thanks for writing us about it (and debugging the issue)

ShayAgros, 2021-10-07:
The fix is now part of [release 2.6](https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.6.0) (this is its [commit](https://github.com/amzn/amzn-drivers/commit/b952df81751938a1912958ae92a465ab96e11548)).

Resolving this issue

closed 2021-10-07 by ShayAgros