amzn/amzn-drivers #171: ENA DPDK driver ena_tx_queue_setup() doesn't honor given socket_id.

soulofmitsui, 2021-05-17:
Hi,

ena_ethdev.c::ena_tx_queue_setup() uses rte_zmalloc() instead of rte_zmalloc_socket() with the given socket_id. Since some of the allocated memory (e.g. tx_buffer_info) are referenced by the fast path, would it be better to force malloc on the given socket_id?

The same also applies to ena_rx_queue_setup().

Thanks!

I-gor-C, 2021-05-17:
Thanks @soulofmitsui, we'll look into this

shaibran, 2021-07-22:
@soulofmitsui Thank you for the observation, you are correct. NUMA adjustments are planned for the DPDK v21.11 release

shaibran, 2021-12-14:
@soulofmitsui  Hey, we up streamed this to DPDK v21.11 (ENA v2.5.0):
http://git.dpdk.org/dpdk/commit/?id=08180833cb698bd737d760e4af77b796882ef328

closed 2021-12-23 by akiyano