amzn/amzn-drivers #172: Promiscuous mode not supported in ENA

kaprakashr, 2021-05-20:
Background
--------------------
We are developing a L2 forwarding switch using AWS_ENA+DPDK.

Issue Description
---------------------
The absence of promiscuous mode makes the implementation hard.
Any packet without targeted MAC should also be supported as a part of a L2 switch functionality

rte_eth_promiscuous_enable() consistently fails for DPDK.

Please clarify!
(This is a bottleneck for many DPDK+AWS implementations as far as i checked with other developers)

Thanks,
Kalyan

I-gor-C, 2021-05-20:
Hi @kaprakashr 
Promiscuous mode is currently not supported by the ENA driver.
We will check if support can be added in the upcoming releases.

shaibran, 2021-12-23:
closing issue (will not fix)

closed 2021-12-23 by akiyano