amzn/amzn-drivers #173: XDP_ZEROCOPY

eugeneia, 2021-05-31:
Howdy!

I'm using ena 2.5.0 on a 5.11 linux kernel and I can't set XDP_ZEROCOPY mode in the bind flags. I get back "operation not supported". 

I'm wondering if this is a problem with my setup or if ena doesn't support zero copy mode?

Any hints would be much appreciated! Thanks!

ShayAgros, 2021-06-01:
Hi, sorry for the late response.

I'm guessing you're using AF XDP (correct me if I'm wrong here). 

The driver doesn't support AF_XDP yet and the application falls back to SKB mode (kernel's generic implementation for XDP processing) which implies copying.

We plan to add AF_XDP support to ENA driver in the upcoming (or the one after it) release.
Please correct me if I got anything wrong in this reply. Otherwise, do you mind sharing your expected use-case for AF_XDP in email shayagr@amazon.com ? It would help us concentrate our efforts on the needs of our users.

Thanks,
Shay

eugeneia, 2021-06-02:
Heyhey no rush! :-)

Yeah you are correct, I’m using AF_XDP. Should have been more clear on that!^^

> We plan to add AF_XDP support to ENA driver in the upcoming (or the one after it) release.

That’s great timing actually! I’d be interested in testing any pre-release versions and giving feedback!

Our use-case is that we have an application that mostly shovels UDP traffic and its bottlenecked by the Linux networking stack. I’m researching ways to accelerate this application by moving the networking into user-space. The way we’d use AF_XFP boils down to:

 - loading an XDP BPF program on the interface that directs UDP traffic destined towards a specific port range to our user-space XSK
 - receiving/transmitting raw ethernet frames via this XSK in our user-space app
 - ideally, we’d get performance close to using AF_XDP with e.g. i40e or other mainstream 10G NICs (i.e., approx. line rate / in the order of millions of packets per second handled on a single core), although anything that bypasses the SKB code paths would probably be a big improvement

Cheers,
Max

eugeneia, 2021-06-02:
Here’s a gist that illustrates the basic XDP setup I’m considering (a UDP source, sink, and a BPF program that matches the ingress traffic we’re interested in, in the real world app source/sink would be a single process).

https://gist.github.com/eugeneia/9eefaf36457000c31de90487654a4299

Ideally we’d be able to run this with

```
    .xdp_flags = XDP_FLAGS_DRV_MODE,
    .bind_flags = XDP_ZEROCOPY | XDP_RING_NEED_WAKEUP,
```

and get good perf.

ShayAgros, 2021-12-08:
> That’s great timing actually! I’d be interested in testing any pre-release versions and giving feedback!

@eugeneia is it still relevant ? We have such pre-release implementation now (: 
If so, please contact me via email shayagr@amazon.com

ShayAgros, 2022-03-30:
Hi, somewhat later than what was anticipated at first, version 2.7.0 of the driver [is out](https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.7.0) with AF XDP zero-copy support. Resolving this issue

closed 2022-03-30 by ShayAgros