amzn/amzn-drivers #175: Issues while upgrading AMI to v1.12.1

afernandezody, 2021-06-14:
Hello,
I'm trying to update the EFA drivers to v1.12.0 or later. Building on an awsparallelcluster AMI (CentOS8 flavor) and after upgrading the kernels to 4.18.0-305.3.1.el8.x86_64, running the full installer returns the following error:
```
= Starting Amazon Elastic Fabric Adapter Installation Script =
= EFA Installer Version: 1.12.1 =

efa-config already installed
== Installing EFA dependencies ==
Main config did not have a skip_missing_names_on_install attr. before setopt
Main config did not have a skip_missing_names_on_install attr. before setopt
Last metadata expiration check: 0:32:03 ago on Mon 14 Jun 2021 03:00:11 PM UTC.
Package kernel-devel-4.18.0-305.3.1.el8.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
Main config did not have a skip_missing_names_on_install attr. before setopt
Main config did not have a skip_missing_names_on_install attr. before setopt
Last metadata expiration check: 0:32:05 ago on Mon 14 Jun 2021 03:00:11 PM UTC.
Package pciutils-3.7.0-1.el8.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
== Installing EFA packages ==
dkms is already installed
Unloading EFA kernel module
Error: No matching Packages to list
Installing RPMS/CENT8/x86_64/rdma-core/ibacm-32.1-1.el8.x86_64.rpm
Main config did not have a skip_missing_names_on_install attr. before setopt
Main config did not have a skip_missing_names_on_install attr. before setopt
Last metadata expiration check: 0:32:22 ago on Mon 14 Jun 2021 03:00:11 PM UTC.
Dependencies resolved.
================================================================================
 Package                     Arch       Version          Repository        Size
================================================================================
Upgrading:
 ibacm                       x86_64     32.1-1.el8       @commandline      85 k
 infiniband-diags            x86_64     32.1-1.el8       @commandline     315 k
 infiniband-diags-compat     x86_64     32.1-1.el8       @commandline      29 k
 libibumad                   x86_64     32.1-1.el8       @commandline      24 k
 libibverbs                  x86_64     32.1-1.el8       @commandline     330 k
 libibverbs-utils            x86_64     32.1-1.el8       @commandline      66 k
 librdmacm                   x86_64     32.1-1.el8       @commandline      68 k
 librdmacm-utils             x86_64     32.1-1.el8       @commandline      91 k
 rdma-core                   x86_64     32.1-1.el8       @commandline      53 k
 rdma-core-devel             x86_64     32.1-1.el8       @commandline     319 k

Transaction Summary
================================================================================
Upgrade  10 Packages

Total size: 1.3 M
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                        1/1 
  Running scriptlet: rdma-core-32.1-1.el8.x86_64                            1/1 
  Upgrading        : rdma-core-32.1-1.el8.x86_64                           1/20 
  Running scriptlet: rdma-core-32.1-1.el8.x86_64                           1/20 
  Upgrading        : libibverbs-32.1-1.el8.x86_64                          2/20 
  Running scriptlet: libibverbs-32.1-1.el8.x86_64                          2/20 
  Upgrading        : libibumad-32.1-1.el8.x86_64                           3/20 
  Running scriptlet: libibumad-32.1-1.el8.x86_64                           3/20 
  Upgrading        : infiniband-diags-32.1-1.el8.x86_64                    4/20 
  Running scriptlet: infiniband-diags-32.1-1.el8.x86_64                    4/20 
  Upgrading        : librdmacm-32.1-1.el8.x86_64                           5/20 
  Running scriptlet: librdmacm-32.1-1.el8.x86_64                           5/20 
  Upgrading        : ibacm-32.1-1.el8.x86_64                               6/20 
[... 276 lines cut ...]
  Verifying        : cmake-rpm-macros-3.18.2-9.el8.noarch                           3/8
  Verifying        : libuv-1:1.40.0-1.el8.x86_64                                    4/8
  Verifying        : efa-config-1.8-1.el8.noarch                                    5/8
  Verifying        : efa-config-1.7-1.el8.noarch                                    6/8
  Verifying        : efa-1.12.1-1.el8.x86_64                                        7/8
  Verifying        : efa-1.10.2-1.el8.x86_64                                        8/8
Installed products updated.

Upgraded:
  efa-1.12.1-1.el8.x86_64                  efa-config-1.8-1.el8.noarch
Installed:
  cmake-3.18.2-9.el8.x86_64                     cmake-data-3.18.2-9.el8.noarch
  cmake-rpm-macros-3.18.2-9.el8.noarch          libuv-1:1.40.0-1.el8.x86_64

Complete!
Unloading EFA kernel module
Reloading EFA kernel module
Minimal installation does not include libfabric, skipping test.
===================================================
EFA installation complete.
- Please logout/login to complete the installation.
===================================================

```
Thanks.

galpress, 2021-06-15:
Hi @afernandezody,
Can you please share the AMI id so I can reproduce the issue?

Thanks.

afernandezody, 2021-06-15:
ami-0598a2f18554d1972

galpress, 2021-06-15:
Thanks, will take a look.

wzamazon, 2021-06-15:
@afernandezody 

The first issue (with regular install) are related to CentOS 8's recent upgrade to CentOS 8.4.

During this upgrade, CentOS8 upgraded hwloc from 1.x to 2.x, which are not binary compatible to each other.

The Open MPI package included in EFA installer is built with hwloc 1.x, because it cannot find hwloc 1.x, the installation failed.

We are actively working to solve this issue.

afernandezody, 2021-06-15:
Hi @wzamazon, 
That makes sense. I guess that retrying w/o updating firstly could be a solution (not sure if I want to follow that route) or maybe upgrading the drivers and, after it, updating the system (not sure if anything would break but there's probably only one way to find out).

galpress, 2021-06-17:
The second issue should now be fixed in the latest release v1.12.2.

I'm resolving this issue, please reopen if you have any more questions.

afernandezody, 2021-06-17:
Hi @**galpress**, 
My installations were using the tar file downloaded from https://efa-installer.amazonaws.com... I guess that the files at _amzn-drivers-efa_linux_1.12.2_ will be integrated into something like _aws-efa-installer-1.12.2.tar.gz_, which at this moment is empty. To install the EFA drivers by themselves, do I simply go to ./amzn-drivers-efa_linux_1.12.2/kernel/linux/efa and use cmake or is there any other intermediate step that I should complete?
Thanks.

galpress, 2021-06-17:
The next installer release should include this fix, so your existing workflow should work fine (that's the preferred way).
If you wish to install the driver without the installer you can generate an rpm by running 'make rpm' in the kernel/linux/efa/rpm directory and installing it using yum/rpm.

afernandezody, 2021-06-17:
Hi @**galpress**, 
I would also prefer to use the installer and would wait if the new release were available today or tomorrow morning. However, I'm in a bit of a hurry as I must prepare an AMI by the end of the week. I tried your suggestion but got the error:
```
sudo yum install efa-1.12.2-1.el8.src.rpm
Last metadata expiration check: 1:17:09 ago on Thu 17 Jun 2021 12:59:46 PM UTC.
Error: Will not install a source rpm package (efa-1.12.2-1.el8.src).
```
I also tried with dnf and using localinstall rather than install but same luck. Maybe it's because it's CentOS8 not ALinux2, not sure.
Thanks.

galpress, 2021-06-17:
Can you please try to install the rpm, not the source rpm?

afernandezody, 2021-06-17:
The rpm was in the x86_64 subdirectory (what was I thinking!). After running yum, the only change that I notice is the creation of the subdirectory `/usr/src/efa-1.12.2` but nothing has changed in the `/opt/amazon` or `/opt/amazon/efa` subdirectories.

galpress, 2021-06-17:
Right, these are directories that are installed by the EFA installer (which you didn't use in this case).
If the rpm installation passed your driver should now be updated, you can verify that by running 'modinfo efa' and making sure the version is 1.12.2g.

afernandezody, 2021-06-17:
You are right as the system is returning:
```
filename:       /lib/modules/4.18.0-305.3.1.el8.x86_64/extra/efa.ko.xz
description:    Elastic Fabric Adapter (EFA)
license:        Dual BSD/GPL
author:         Amazon.com, Inc. or its affiliates
softdep:        pre: ib_uverbs
version:        1.12.2g
rhelversion:    8.4
srcversion:     CECDE2333322F004E8B5352
alias:          pci:v00001D0Fd0000EFA1sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EFA0sv*sd*bc*sc*i*
depends:        ib_core,ib_uverbs
name:           efa
vermagic:       4.18.0-305.3.1.el8.x86_64 SMP mod_unload modversions
```
The whole thing is to overcome the issue discussed at https://github.com/ofiwg/libfabric/issues/6332. Hopefully installing the new driver will suffice. Thanks.

galpress, 2021-06-17:
Glad to hear it worked!
Though I'm not sure I understand how updating the driver is related to this issue?

afernandezody, 2021-06-17:
Are you saying that the issue can only be fixed if the whole installer (v1.12) is run?

wzamazon, 2021-06-17:
@afernandezody 

Running whole installer 1.12.x is ideal. Because that is not possible now on CentOS 8, you can install the libfabric-aws-xxx.rpm that comes with installer, which should also fix the fork issue.

afernandezody, 2021-06-17:
@**wzamazon**, 
I downloaded https://efa-installer.amazonaws.com/aws-efa-installer-1.12.1.tar.gz but the rpms in the `./aws-efa-installer/RPMS/CENT8/x86_64` subdirectory are _libfabric-aws-1.11.2amzn1.1-1.el8.x86_64.rpm_ and _libfabric-aws-devel-1.11.2amzn1.1-1.el8.x86_64.rpm_. When I try to install the 1st one, the system states that it's already installed and makes no change (as far as I can see). Shouldn't the file be named something like _libfabric-aws-1.12.1amznX.X-X.el8.x86_64.rpm_? (it's the same for the other OSs)

wzamazon, 2021-06-17:
@afernandezody

EFA installer contains multiple software packages: efa kernel module, lifabric, open mpi, rdma-core. Each component has its own version. So EFA installer version does not necessary match libfabric version.

If you already have  libfabric-aws-1.11.2amzn1.1-1.el8.x86_64 installed on you machine, this should be enough to  address the fork issue.

afernandezody, 2021-06-17:
OK. It will take me today and most of tomorrow until I have a configuration ready to test.

galpress, 2021-06-27:
@afernandezody any updates? Anything else needed from our side?

afernandezody, 2021-06-27:
@**galpress**, 
Sorry, one issue led to another and, at the end, I decided to wait until the release of parallelcluster 1.11.0 to recheck everything out and make any upgrades if necessary. I'm closing the thread (and will only reopen it if the issue reproduces with the newer AMI).

closed 2021-06-27 by afernandezody