amzn/amzn-drivers #178: CPU usage in ENA DPDK driver tx path is much higher on m5dn.8xlarge instances than m5dn.2xlarge instances

soulofmitsui, 2021-07-08:
Hi,

I've written a DPDK app using ENA dpdk driver. It runs well on m5dn.2xlarge instances with very low CPU usage that I'm happy with. However, when I run the same app on m5dn.8xlarge instances, the CPU usage goes much higher surprisingly with the same network b/w (i.e. 25Gb/s). perf top showed eth_ena_xmit_pkts() and ena_com_prepare_tx() are consuming much higher CPU than running on m5dn.2xlarge instances.

 32.36%  librte_net_ena.so.21.0  [.] eth_ena_xmit_pkts
 21.06%  librte_net_ena.so.21.0  [.] ena_com_prepare_tx

 Any idea on what might have caused the high CPU usage? My app creates 32 queues on m5dn.8xlarge instances vs 8 queues on m5dn.2xlarge instances. FWIW I'm using the igb_uio kernel driver with wc_activate=1.

nafeabshara, 2021-07-08:
thanks for raising this issue

how many concurrent flows are you using on the 8xl instance ?  and are you
using more CPU cores ?

the higher CPU load can happen with the CPU generating more than the
network can transmit,  so the drive end up polling a lot waiting for empty
queue and will do less batching on transmit.

soulofmitsui, 2021-07-08:
I've tried doing the same number of current flows with 2xl instances. The CPU usage is still much higher than 2xl instances.

zorikm, 2021-07-09:
@soulofmitsui 
In your application does each CPU core handle single ENA queue? or alternatively, you have a subset of CPU cores, where each of them processes multiple ENA queues?

I-gor-C, 2021-07-09:
@soulofmitsui How does your application perform the congestion control? Does it slow down the tx for a specific queue if driver pushed back?

soulofmitsui, 2021-07-09:
@zorikm, each CPU core handles a single ENA queue (one to one map).
@I-gor-C, yes, or packet drops on network.

I think I found the issue. It's related to the value of ena_dev->tx_mem_queue_type. On 8xlarge, the default value is ENA_ADMIN_PLACEMENT_POLICY_DEV. If I force it to be ENA_ADMIN_PLACEMENT_POLICY_HOST (by hard code and recompile ENA dpdk driver), my CPU usage went back to normal (2-3x lower than ENA_ADMIN_PLACEMENT_POLICY_DEV). I didn't check the default value for 2xlarge. My guess is it would be ENA_ADMIN_PLACEMENT_POLICY_HOST.

Does the above ring a bell? Should I force tx_mem_queue_type to be ENA_ADMIN_PLACEMENT_POLICY_HOST? FWIW, I'm running my app in a k8s container if that changes things.

I-gor-C, 2021-07-09:
POLICY_DEV should be the default for both instance types. 
This could be an issue with WC enablement. Have you followed the instructions in the documentation? Specifically, please use the specific `igb_uio` module that is referenced in the documentation (https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk).

soulofmitsui, 2021-07-09:
Thanks @I-gor-C. You are correct. I used the wrong kmods from https://github.com/atsgen/dpdk-kmod (shows up first in google). I must have used the correct kmods on 2xl.

Really appreciate your help!

I-gor-C, 2021-07-09:
Glad it helped.
Closing this issue, please reopen if further assistance is required.

closed 2021-07-09 by I-gor-C

zorikm, 2021-07-09:
PLACEMENT_POLICY_DEV should be a default setting for both. It means that the Tx ring is located in the device memory and CPU "pushes" there descriptors and packet headers. Worth mentioning that the ring area is mapped as write-combine to guarantee cache line sized writes over PCI. In this mode indeed CPU works a little harder but this reduces DMA reads and overall improves latency and PPS.
PLACEMENT_POLICY_HOST is when Tx ring is located in the instance memory, and descriptors and headers are DMAed by the card.
PLACEMENT_POLICY_DEV is a default configuration and should be the same also for 2xl.

Is per Tx ring traffic volume similar for  2xl and 8xl. Are CPU utilization numbers you mentioned above consistent across all cores, or it's an average one? What 2xl CPU utilization range have you observed?