amzn/amzn-drivers #179: DPDK App gets only 50Gb/s on m5dn.24xlarge instances

soulofmitsui, 2021-07-10:
Hi,

I'm not sure if this is DPDK driver related - apology if not.

I'm testing my DPDK app on m5dn.24xlarge instances which are quoted with 100Gb/s b/w. However I'm only able to get 50Gb/s. I've tried different flow count over one or multiple ENA interfaces which all have the same result. There are plenty CPU free everywhere when b/w caps at 50Gb/s.

I also tried iperf using TCP which is able to reach 100Gb/s with multiple flows. My app uses UDP though.

Since the b/w seems to cap at exactly 50Gb/s, it feels something is artificially throttling. One wild guess I have is the ENA interface is assumed to be attached to numa node 0 (numa_node is set to -1 but DPDK driver assumes it's 0 by default) which limited the b/w to half on two numa nodes instance.

Is there anything else could artificially cap the b/w? Any help is much appreciated!

I-gor-C, 2021-07-10:
Hi @soulofmitsui 
Could you please provide your instance id and reproduce the issue?

soulofmitsui, 2021-07-10:
Thanks @I-gor-C. Just sent.

I-gor-C, 2021-07-10:
Do you utilize all 32 queues and have similar number of flows per queue?

soulofmitsui, 2021-07-10:
yes, all 32 queues with 1 flow per queue.

I-gor-C, 2021-07-21:
The issue was resolved, closing.

closed 2021-07-21 by I-gor-C

alan-copeland-keysight, 2021-11-07:
What was the issue? I think I am seeing something similar (capped at 50 Gbps, UDP traffic)

soulofmitsui, 2021-11-07:
@alan-copeland-keysight, the issue was caused not calling rte_eth_tx_prepare() which is a must if you use h/w chksum offload.

shaibran, 2021-11-10:
@alan-copeland-keysight  was your issue resolved or do you need further assistance?

alan-copeland-keysight, 2021-11-12:
Sorry, I was pulled off on to another task. I'm hopeful this will help, and I'll update when I get a chance to try it.