amzn/amzn-drivers #180: Debian OS support for EFA?

davidshtian, 2021-08-02:
Based on the AWS EC2 guide (https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/efa.html#efa-amis), only following AMIs support EFA with Intel x86-based instance types:

- Amazon Linux 2
- CentOS 7 and 8
- RHEL 7 and 8
- Ubuntu 18.04 and 20.04
- SUSE Linux Enterprise 15 SP2 and later
- openSUSE Leap 15.2 and later

Is there any plan to support AMI like Debian? And what's the restriction? Thanks~

sleybo, 2021-08-04:
Hi 

Currently we don't have plans to support Debian. 
We will review the required effort to support it and update.

Thanks
Yossi

cordeezy2, 2024-05-25:
Any update to this? Any plans to support Debian? Any specific Restriction ?

YonatanNachum, 2024-12-23:
Hey, EFA is indeed supported on Debian as can be seen in: [EFA supported operating systems](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/efa.html#efa-os).

Resolving this issue.

closed 2024-12-23 by YonatanNachum