amzn/amzn-drivers #181: can not install driver upon linux boot

romange, 2021-08-10:
os: Ubuntu 21.04

I do all the steps as described in the documentation:
1.  `echo ena | sudo tee /etc/modules-load.d/ena.conf`
2.  `sudo make -C /lib/modules/`uname -r`/build M=`pwd` modules_install`
3.  `sudo depmod` and reboot

I also had to add a signing key for step 2 to succeed:

```
cd /lib/modules/$(uname -r)/build/certs

sudo tee x509.genkey > /dev/null << 'EOF'
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts
[ req_distinguished_name ]
CN = Modules
[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid
EOF

sudo openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem
```

After all these steps ena module has not been updated after the reboot.

akiyano, 2021-08-10:
Hi @romange ,

Please try the following steps to install the driver using the DKMS utility:

1. sudo apt-get install dkms
2. git clone git@github.com:amzn/amzn-drivers.git
3. sudo mv amzn-drivers /usr/src/amzn-drivers-2.5.0

4. create the file /usr/src/amzn-drivers-2.5.0/dkms.conf with the following content:
```
PACKAGE_NAME="ena"
PACKAGE_VERSION="1.0.0"
CLEAN="make -C kernel/linux/ena clean"
MAKE="make -C kernel/linux/ena/ BUILD_KERNEL=${kernelver}"
BUILT_MODULE_NAME[0]="ena"
BUILT_MODULE_LOCATION="kernel/linux/ena"
DEST_MODULE_LOCATION[0]="/updates"
DEST_MODULE_NAME[0]="ena"
AUTOINSTALL="yes"
```

5. sudo dkms add -m amzn-drivers -v 2.5.0
6. sudo dkms build -m amzn-drivers -v 2.5.0
7. sudo dkms install -m amzn-drivers -v 2.5.0
8. reboot the instance

Thanks,
Arthur

romange, 2021-08-11:
Works, like a charm. Thanks a lot!
Suggestions: add this recipe to ena README.

closed 2021-08-11 by romange