amzn/amzn-drivers #184: RHEL8 support using CloudWatch agent ethtool metrics

lazize, 2021-09-08:
Hi,

I am facing issues to use metrics from ENA on RHEL8.

I have CloudWatch agent configured to get metrics from ethtool as described on link below. The difference is that I am getting it from `eth0`.  
https://docs.aws.amazon.com/AmazonCloudWatch/latest/monitoring/CloudWatch-Agent-network-performance.html

According to the link above ENA driver version should be 2.2.10 or later. But it doesn't show me in the same way as documented.

```shell
[root@ip-10-91-0-51 ~]# ethtool -i eth0 | grep version
version: 4.18.0-305.12.1.el8_4.x86_64
firmware-version:
expansion-rom-version:

```

Also I can't see any `exceeded` metric from ethtool on instance console
```shell
[root@ip-10-91-0-51 ~]# ethtool -S eth0 | grep exceeded
```

More details about my instance:

```shell
[root@ip-10-91-0-51 ~]# modinfo ena
filename:       /lib/modules/4.18.0-305.12.1.el8_4.x86_64/kernel/drivers/net/ethernet/amazon/ena/ena.ko.xz
version:        2.1.0K
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
rhelversion:    8.4
srcversion:     442F0C70CBBBBA7715FB41C
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000051sv*sd*bc*sc*i*
depends:
intree:         Y
name:           ena
vermagic:       4.18.0-305.12.1.el8_4.x86_64 SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Red Hat Enterprise Linux kernel signing key
sig_key:        65:AE:C8:FF:D2:B5:69:16:C0:05:18:8A:66:4D:C4:A4:44:76:87:56
sig_hashalgo:   sha256
signature:      AA:BD:F7:0C:B7:84:08:DE:86:66:4C:E8:70:F3:CF:79:E0:55:87:BA:
                0F:6F:85:53:D6:AC:45:E0:4D:57:F2:49:73:70:98:96:06:9A:74:81:
                50:A8:E5:F7:EA:D4:94:21:D1:8D:EA:27:8E:20:65:05:17:4F:98:FB:
                C2:49:FD:78:06:54:28:14:1D:C1:5E:20:7F:3D:B7:E4:C4:DA:B8:3F:
                DE:9A:AD:91:A0:93:E8:DF:98:34:D0:8C:17:BA:9E:21:90:E3:87:62:
                B5:80:B6:F4:DB:9D:0A:87:1F:2A:0C:41:06:CE:EA:E0:BF:8E:33:E8:
                5B:38:78:FA:BB:44:9D:8D:FF:09:56:15:8A:43:D2:84:37:95:61:98:
                48:C0:D1:19:96:96:04:D5:5D:02:B5:AA:8B:90:46:14:3D:25:25:49:
                49:DC:CB:3B:F2:8B:D6:D6:93:FD:DB:96:4E:29:CF:66:67:37:FC:B7:
                7F:AC:F2:BA:99:5B:25:D2:50:12:4E:F7:B7:2B:A8:CF:BD:A5:44:7B:
                34:BA:21:84:CB:C7:F2:43:85:3E:74:2A:31:B4:BF:F6:6A:DD:93:9A:
                BA:13:73:18:88:88:53:FE:1B:84:C6:5A:FE:6B:35:79:22:5D:95:78:
                F9:2F:A3:8A:9A:7B:11:AE:5C:4D:23:49:68:0A:76:DE:2F:BF:18:06:
                B8:E9:00:8D:B0:AD:B3:B1:4F:5E:D7:3F:2C:94:07:B8:F8:52:74:9C:
                2E:C1:3D:9D:2C:97:85:10:B7:8F:4F:F2:EE:3A:92:80:10:92:78:58:
                54:27:98:F9:A7:24:42:23:0C:41:43:65:4C:D2:D3:45:BD:B3:64:BD:
                AE:F2:AB:0E:F8:FA:C9:35:DD:CC:D1:12:C8:37:FF:9B:E1:5C:46:B9:
                7C:BA:2D:BC:87:45:C2:DE:1C:F9:27:47:71:68:4E:B0:BA:87:B9:58:
                78:D1:C1:A4:56:38:64:70:4F:4D:A2:03:32:25:43:4D:7A:A5:42:9C:
                45:E9:B6:85
parm:           debug:Debug level (0=none,...,16=all) (int)
```

zorikm, 2021-09-08:
Hi Leonardo,
Regarding your first question about the driver version:
Linux folks has decided that driver version should not be published, and that kernel version is sufficient. Therefore our upstream ENA driver as well as ENA driver included in major distros has to follow this guidance.
You are welcome to use our GitHub ENA driver, as well as the driver included in Amazon Linux 2 that still expose driver version.
   
Regarding your second question:
It indeed looks like the driver included in  RHEL8.4 (4.18.0-305.el8.x86_64) doesn't support the new network performance metrics. Please let us investigate and get back to you. In the meantime to enable collection of these metrics you can install our latest GitHub ENA driver.
Thanks,
Zorik

zorikm, 2021-09-09:
Hi Leonardo,
According to the information we have, the metrics you are looking for will be supported in RHEL 8.5

Thanks,
Zorik

lazize, 2021-09-10:
Thank you for your help @zorikm

I can confirm that upgrading the driver with github version fixed the issue.

closed 2021-09-29 by akiyano