amzn/amzn-drivers #185: Linux ENA 2.5.0 pps throttling based on queue instead of instance

ricebus, 2021-09-12:
Hi,

We’ve upgraded our driver version to 2.5.0 and done some performance tests and it seems there is a different behavior in the new driver with concern to queue count / throttling (compared to 1.5.0)

Instance for all tests: c5n.xlarge
Interface count: 2 (this is a FW, so 1 external 1 internal)

Test 1:
Driver: 1.5.0
Queue count: 4 (unchangeable)
Affinity: 2 on core0, 2 on core2 (per interface)
Throughput:  ~7Gbps

Test 2:
Driver: 2.5.0
Queue count: 2
Affinity: core0, core2
Throughput: ~5.8Gbps

Test 3:
Driver: 2.5.0
Queue count: 4
Affinity: 2 on core 0, 2 on core 2 (per interface)
Throughput:~7Gbps

When going through test 2 we used ethtool –S and saw pps_allowance_exceeded going up and so there were a lot of drops.
And when we changed the queue amount in test 3 the number stopped incrementing so dramatically (but still got a few of them)

Per this blog post: https://aws.amazon.com/blogs/networking-and-content-delivery/amazon-ec2-instance-level-network-performance-metrics-uncover-new-insights/

“**pps_allowance_exceeded** indicates number of packets shaped and dropped due to Packet Per Second (PPS) exceeding the allowance for the **instance**.
PPS allowance is enforced separately to the overall bandwidth allowance and while the instance may still be under overall bandwidth allowance the PPS allowance may exceed if the mean packet size is small.”

If this is an instance limit, why is it enforced on a queue basis? This looks like a problem.

ShayAgros, 2021-09-13:
Hi, thanks for reaching out

> If this is an instance limit, why is it enforced on a queue basis? This looks like a problem.

I'm not aware of this limit being per-queue basis, but perhaps you observed a bug.

Bringing up the following setup
```
c5n.18xlarge      c5n.xlarge
┌────────┐        ┌────────┐
│        │        │        │
│   ┌────┤9K TCP  ├─────┐  │
│   │    ├───────►│     │  │
│   │eth1│        │eth1 │  │
│   │32Qs│        │2 Qs │  │
│   └────┤        ├─────┘  │
│        │        │        │
└────────┘        └────────┘
```

I manage to reach 25Gbps and 339Kpps.

When sending 64byte traffic instead I manage to reach 1.8Mpps.
Both the bandwidth and pps correspond to the max limit values I'm aware of.
The traffic is sent using [iperf2](https://sourceforge.net/projects/iperf2/) tool. Also I set the affinity the way you described.

Did you use c5n.xlarge for both RX and TX ?

ricebus, 2021-09-20:
Yes, the machine is used for forwarding/routing so maybe you need to test it with a TX/RX scenario.
We're using TCP ~1500 bytes traffic (no jumbo frames).

ricebus, 2021-10-03:
Hi, any news?

zorikm, 2021-10-04:
Hi @ricebus 
Apologies for the delay. 
By reducing the number of queues while keeping a similar traffic volume, you contribute to uneven load distribution across device resources assigned to your instance. As a result, some of the resources get overloaded, and that's where you hit allowance thresholds. 
This serves as an indicator - for such traffic load you might consider to increase the number of queues.
Thanks.

ricebus, 2021-10-05:
@zorikm 
So the documentation needs to be adjusted accordingly. maybe state that the PPS limit is for all available queues meaning that when you reduce the amount of queues you revoke some of your allocated resources.

zorikm, 2021-10-05:
@ricebus ,
Thank you for your comments. 
Agreed, we will update our documentation.

zeffron, 2022-10-19:
Does this mean that for XDP, only ~50% of the pps is available because XDP can only use half of the queues? See also #240.

davidarinzon, 2022-10-20:
Hi @zeffron 

Regarding your question:
> Does this mean that for XDP, only ~50% of the pps is available because XDP can only use half of the queues? See also #240.
Please note the answer provided in https://github.com/amzn/amzn-drivers/issues/185#issuecomment-933883165, reducing the amount of queues potentially impacts performance, but it depends on your workloads and configuration.

davidarinzon, 2022-10-23:
Hi @zeffron 

To clarify regarding your previous question and response https://github.com/amzn/amzn-drivers/issues/185#issuecomment-1285393550, if your question was focused on device side limitations, there are no special limitations for XDP traffic.

Thanks!

zeffron, 2023-10-16:
@davidarinzon Sorry it took me a year to respond. My question is not about device side limitations, I don't think. Let me see if I can clarify.
In https://github.com/amzn/amzn-drivers/issues/185#issuecomment-933883165 @zorikm said
> By reducing the number of queues while keeping a similar traffic volume, you contribute to uneven load distribution across device resources assigned to your instance. As a result, some of the resources get overloaded, and that's where you hit allowance thresholds.

XDP on ENA, as a requirement, reduces the number of queues by 50% (see https://github.com/amzn/amzn-drivers/issues/240).

Putting those two pieces of information together, it seems like XDP on ENA would result in some resources getting overloaded _that would not if XDP did not require cutting the queues in half_. Is that an accurate understanding?

davidarinzon, 2024-04-07:
Hi @zeffron 
Apologies, I noticed that your comment was left unanswered.
You are correct, in XDP use-case, where the number of queues are reduced by 50%, you might experience potential performance impact.
Have you observed such in your XDP workloads?

davidarinzon, 2024-04-07:
In regard to this issue and the documentation, you may consult with https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ena-nitro-perf.html, which also mentions the impact of number of queues reduction.

closed 2024-04-24 by davidarinzon