amzn/amzn-drivers #186: Debian 10.10 Os Support issue for ENA (after Debian migration from 8.9 to 10.10)

Kumoservices, 2021-10-05:
Hello,
In EC2, we have migrate OS on a old t2.small instance from debian 8.9 to 10.10, to goes to t3 instance type of m5 instance type in target.

When i run this nitro_check_script.sh  :

```
OK     NVMe Module is installed and available on your instance

ERROR  ENA Module is not available on your instance. 
	- Please install ENA module before changing your instance type to Nitro. Look at the following link for further guidance:
	> https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html#enhanced-networking-ena-linux

OK     fstab file looks fine and does not contain any device names.
```

So i need to setup the ENA drivers on has nitro script said that module is not in.

When i make the ENA linux amazon drivers i have this error :
```

~/amzn-drivers-master/kernel/linux/ena # make
make -C /lib/modules/3.16.0-4-amd64/build M=/root/amzn-drivers-master/kernel/linux/ena modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-4-amd64'
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-amd64'
  CC [M]  /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.o
In file included from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.h:9:0,
                 from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:25:
/root/amzn-drivers-master/kernel/linux/ena/kcompat.h:717:0: warning: "net_get_random_once" redefined [enabled by default]
 #define net_get_random_once(buf, nbytes)         \
 ^
In file included from /usr/src/linux-headers-3.16.0-4-common/include/linux/skbuff.h:27:0,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/if_ether.h:23,
                 from /usr/src/linux-headers-3.16.0-4-common/include/uapi/linux/ethtool.h:17,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/ethtool.h:16,
                 from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:11:
/usr/src/linux-headers-3.16.0-4-common/include/linux/net.h:251:0: note: this is the location of the previous definition
 #define net_get_random_once(buf, nbytes)    \
 ^
In file included from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.h:9:0,
                 from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:25:
/root/amzn-drivers-master/kernel/linux/ena/kcompat.h:724:20: error: redefinition of 'netdev_rss_key_fill'
 static inline void netdev_rss_key_fill(void *buffer, size_t len)
                    ^
compilation terminated due to -Wfatal-errors.
make[4]: *** [/usr/src/linux-headers-3.16.0-4-common/scripts/Makefile.build:263: /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.o] Error 1
make[3]: *** [/usr/src/linux-headers-3.16.0-4-common/Makefile:1354: _module_/root/amzn-drivers-master/kernel/linux/ena] Error 2
make[2]: *** [Makefile:182: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-4-amd64'
make: *** [Makefile:40: all] Error 2
```

```
 ~/amzn-drivers-master/kernel/linux/ena # uname -a && cat /etc/debian_version
Linux My_EC2_instance #1 SMP Debian 3.16.43-2+deb8u2 (2017-06-26) x86_64 GNU/Linux
10.10
```

I have open a case in aws support, but without solution at once (case id : 8980175661)

regards

Kumoservices, 2021-10-05:
I have try a OS update from 8.9 to 9.13, and i have same sort of issue :
```
root@clicrdv.aws-eu-west-01.mongodbpp-03 ~/amzn-drivers-master/kernel/linux/ena # make
make -C /lib/modules/3.16.0-4-amd64/build M=/root/amzn-drivers-master/kernel/linux/ena modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-4-amd64'
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-amd64'
  CC [M]  /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.o
In file included from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.h:9:0,
                 from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:25:
/root/amzn-drivers-master/kernel/linux/ena/kcompat.h:717:0: warning: "net_get_random_once" redefined [enabled by default]
 #define net_get_random_once(buf, nbytes)         \
 ^
In file included from /usr/src/linux-headers-3.16.0-4-common/include/linux/skbuff.h:27:0,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/if_ether.h:23,
                 from /usr/src/linux-headers-3.16.0-4-common/include/uapi/linux/ethtool.h:17,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/ethtool.h:16,
                 from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:11:
/usr/src/linux-headers-3.16.0-4-common/include/linux/net.h:251:0: note: this is the location of the previous definition
 #define net_get_random_once(buf, nbytes)    \
 ^
In file included from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.h:9:0,
                 from /root/amzn-drivers-master/kernel/linux/ena/ena_netdev.c:25:
/root/amzn-drivers-master/kernel/linux/ena/kcompat.h:724:20: error: redefinition of 'netdev_rss_key_fill'
 static inline void netdev_rss_key_fill(void *buffer, size_t len)
                    ^
compilation terminated due to -Wfatal-errors.
/usr/src/linux-headers-3.16.0-4-common/scripts/Makefile.build:262: recipe for target '/root/amzn-drivers-master/kernel/linux/ena/ena_netdev.o' failed
make[4]: *** [/root/amzn-drivers-master/kernel/linux/ena/ena_netdev.o] Error 1
/usr/src/linux-headers-3.16.0-4-common/Makefile:1354: recipe for target '_module_/root/amzn-drivers-master/kernel/linux/ena' failed
make[3]: *** [_module_/root/amzn-drivers-master/kernel/linux/ena] Error 2
Makefile:181: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-4-amd64'
Makefile:40: recipe for target 'all' failed
make: *** [all] Error 2
```

akiyano, 2021-10-05:
Hi @Kumoservices,

Regarding Debian 8:
I was able to reproduce your compilation error.

Regarding Debian 9:
When I ran a clean Debian 9 instance (debian-stretch-hvm-x86_64-gp2-2021-06-01-74413 (ami-07a2fd1dc58356f1e) in eu-west-1 region), the ena driver compiled without errors.
Looking at the output you posted for Debian 9, I see that the compilation is done with the headers of kernel 3.16, which is the kernel version of Debian 8 and not of Debian 9.
My guess is that you did not reboot your instance after upgrading to Debian 9. 
After the reboot please run the command “uname -r” and it should return kernel version 4.9.X.

Please try rebooting the instance and see if it solves your issue. 

Regards,
Arthur

NethanelK, 2021-11-28:
Hi @Kumoservices,

Sorry it took some time
I attach a patch which should fix this compilation issue in debian 8.
You can apply it with:
`git am 0001-linux-ena-Fix-compilation-error-in-debian-8.txt`

The patch would also be included in next driver release

[0001-linux-ena-Fix-compilation-error-in-debian-8.txt](https://github.com/amzn/amzn-drivers/files/7613429/0001-linux-ena-Fix-compilation-error-in-debian-8.txt)

Please let me know if this patch fixes your issue
Nati

akiyano, 2022-03-30:
Driver 2.7.0g, which handles this issue in debian 8,  was released.
Closing the issue.

closed 2022-03-30 by akiyano